Ultimaker/Cura #8065: Implement variable line width.

Sophist-UK, 2020-07-14:
**Is your feature request related to a problem? Please describe.**
https://www.facebook.com/groups/1959877270949115/permalink/2699062390363929/

**Describe the solution you'd like**
Provide settings for lower and upper % of nozzle width for inner and outer walls and allow Cura to vary the line width dynamically in order to fully fill narrow walls. For example suppose nozzle width is .4mm and you want to print a 1mm wall - Cura would then print 2 outer walls each with a line width of 125%. Similarly a wall of 1.1mm might be printed with outer walls at .4mm and a .3mm inner wall. A 1.8mm wall might be printed with outer walls at 0.4mm and two inner walls at .5mm.

**Describe alternatives you've considered**
None.

**Affected users and/or printers**
People producing engineering objects where strength of thin walls are important enough to want a good solid fill.

fieldOfView, 2020-07-15:
Could you please post the issue _here_? I am not allowed to even read the page without logging in to Facebook and joining that group.

rburema, 2020-07-15:
We're working on something similar [here](https://github.com/Ultimaker/CuraEngine/pull/1210) (and related branches). It's more of an automated line width solution, though. I'm not sure if (when all is said & done) the user will supply a min/max width themselves, or if that will be in the config or even just not configurable.

Sophist-UK, 2020-07-15:
Sorry - here is the facebook post anonymised:

Д: Why Cura does this? small extrudes and a lot of movements
[image]
> St: In general your line width should be the same as your nozzle size, so set them to 0.4

>> Д: I use 1mm nozzle. And because of nozzle design its fine to slightly increase line width.

>> St: I see, did not realise you were using a large nozzle. Try turning on " print thin walls" in cura. And maybe turn down your inner wall line width, maybe cura can fill it in with a line then.

>> Д: Switching on or off "print thin walls" setting doesn't make any change to this :( Its a difficult place to play with line width, because shape continuously increased , like trapezoidal. So this weird dotted lines just move below or above this layer.

>> Д: If you interested I will publish stl to inspect. But tomorrow. Now time to sleep

>> St: Not sure how your going to do it, but you maybe able to remove all the dots ( but the model will loose strength) try setting " fill gaps between walls" to nowhere. sorry not sure what else to suggest. Have a good night :)

>> Д: Already do this :) not helps.

>> Sophist-UK: There is nothing wrong with using a line width greater than the nozzle size. It extrudes more filament and squashes it sideways leading to greater layer adhesion. (See CNC Kitchen on YouTube.} Use on inner walls.

>> Д: I say and do exactly the same :)

>> Sophist-UK: Unfortunately, whilst Cura has a variable layer height option (never used it) it doesn't have a variable line width option. Which is a pity, because it would be ideal for this type of situation. In essence, you would print outer walls first, and then Cura would adjust the line width between (say) 80% and 150% of the nozzle width in order to fill the gap completely.

>> Д: Yeah, this function can solve some slicing issues

rburema, 2020-07-15:
Ah, that image clears up something.

While the 'libArachne' fix will still be the eventual 'true' fix, in this case, there may be a workaround, since it _could be_ rounding errors that cause this. That is, if the width of the feature is very close to the line-width (if I 'read' the screenshot correctly). The workaround (in that case) is to either make the line-width thinner by just a _very tiny_ amount, or to scale the model up by a _very tiny_ amount.

Sophist-UK, 2020-07-15:
@rburema said: "The workaround (in that case) is to either make the line-width thinner by just a very tiny amount, or to scale the model up by a very tiny amount."

Neither will work in the example provided because the walls are tapering - and get gradually wider (or narrower) layer by layer.

Ghostkeeper, 2020-07-16:
I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/7687. See also my explanation of why this happens here: https://github.com/Ultimaker/Cura/issues/7687#issuecomment-625553512

Reducing the inner wall line width by 0.02 or so usually works for me as a workaround. It wouldn't be visible on the outside. Our "true" fix is in the works here: https://github.com/Ultimaker/CuraEngine/pull/1210

closed 2020-07-16 by Ghostkeeper