Ultimaker/Cura #8067: Project model moved when reloaded

Sophist-UK, 2020-07-15:
**Application version**
4.60

**Platform**
Windows 10

**Printer**
Dagoma DiscoEasy200

**Reproduction steps**
1. I have a model that occupies the full width of the build plate. The model position is centred i.e. x=0, y=0. I saved it as a project.
2. I closed Cura and reopened it and loaded the project I just saved. Model was now at x=-2.9518, y=3.8244.

I get this consistently, It is obviously easy to reset x and y, but still a little annoying,

**Screenshot(s)**
Not needed.

**Actual results**
Model not loaded at expected position.

**Expected results**
Model loads at expected position.

**Project file**
[phaser_side2_repaired.zip](https://github.com/Ultimaker/Cura/files/4924808/phaser_side2_repaired.zip)

**Log file**
[xy.log](https://github.com/Ultimaker/Cura/files/4924830/xy.log)

mahtDFR, 2020-07-16:
Can't seem to reproduce this over here, it's opening at position 0,0 for me.
[image]

Sophist-UK, 2020-07-16:
I am assuming it is a printer definition that is causing it. Can you load the Dagoma DiscoEasy profile and try that?

mahtDFR, 2020-07-16:
This is the same definition as you (DE200), it's loaded from the 3mf.

Sophist-UK, 2020-07-16:
I suppose it could be 4.6.0 vs. 4.6.1 - but I do get this issue consistently. I guess we will just have to write this off as a minor inconvenience.

closed 2020-07-16 by Sophist-UK