Ultimaker/Cura #8070: Nozzle not changing Diameter

BunnyTJ, 2020-07-16:
When I go to printer settings and change my nozzle diameter it does not change to the desired Diameter, as soon as I say close it reverts back to 0.4mm

mahtDFR, 2020-07-16:
Please fill in the issue template so we are able to help you:

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-07-17:
Does it still revert if you enter the nozzle diameter with a comma `0,8` instead of a period `0.8`?

BunnyTJ, 2020-07-19:
Hi. It stays in place now is that then the issue? It should be a comma?

Ghostkeeper, 2020-07-21:
The issue is that it depends on your locale. It's in fact the same bug then as what [you yourself opened before](https://github.com/Ultimaker/Cura/issues/6546). I'll close this as a duplicate.

closed 2020-07-21 by Ghostkeeper