Ultimaker/Cura #8072: [4.6.1] Print Speed is not saved in new material profile

buteomont, 2020-07-16:
**Application version**
4.6.1

**Platform**
Ubuntu Linux Bionic Beaver

**Printer**
Custom Hictop 3DP-18

**Reproduction steps**

1. Create a new material profile
2. Export the profile and edit the *.xml.fdm_material file
3. Add a line under `<settings>` section: `<setting key="print speed">30</setting>`
4. Import the material profile into Cura materials (no errors noted)
5. Select the material for use.  Print speed does not change.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Print speed is not affected when new material is loaded, and no error is produced when profile is imported into cura

**Expected results**
Print speed should be that which was specified in the material profile -OR- an error should appear during import of profile

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-07-20:
It could very well be that a quality profile also sets the print speed. As such, it will override whatever is set in the material profile. See https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings for more details how the profile system work.

If you want to see where cura gets the values from, I would recomend installing the godmode plugin (https://github.com/sedwards2009/cura-god-mode-plugin).

Could you check if this is indeed the case?

Ghostkeeper, 2020-07-20:
Also `print speed` is not a known setting key. Use this instead:

    <cura:setting key="speed_print">30</cura:setting>

And make sure that the `cura` namespace is defined at the top of the file using `xmlns:cura="http://www.ultimaker.com/cura"`.

buteomont, 2020-07-20:
> Also `print speed` is not a known setting key. Use this instead:
> ```
> <cura:setting key="speed_print">30</cura:setting>
> ```

This fixed the problem, thanks.
This leaves me only three questions:

1. Shouldn't my incorrect setting key have given an error during import?
2. Where can one find a list of valid key names?
3. Is there a reason that only a subset of settings are available in the "manage materials" GUI?

fieldOfView, 2020-07-21:
You may be interested in the Material Settings plugin from the Marketplace, which will let you set almost all settings in the material.
https://github.com/fieldOfView/Cura-MaterialSettingsPlugin/

nallath, 2020-07-21:
1. Nope. Cura isn't the only application that uses these files. So we can't be too strict about it.
2. All the settings as defined in fdmprinter.def.json
3. See the plugin from fieldofview. The reason why we didn't do it is mostly legacy and it hasn't been changed since.

closed 2020-07-21 by nallath