Ultimaker/Cura #8079: Missing slide bar from preview on bottom of screen [4.6.1] and earlier versions. (4.x.x)

bteamfox, 2020-07-17:
**Application version**
4.6.1 and earlier versions. (4.x.x)

**Platform**
Acer Extensa 5235 Laptop
CPU: Pentium T4500 (2) @ 2.300GHz
GPU: Intel Mobile 4 Series Chipset
Resolution: 1366x768
Linux 19.2, 19.3, 20

**Printer**
Ender-3

**Reproduction steps**
Slice any model then click preview.

**Screenshot(s)**
[image]

**Actual results**
There is no bar near the bottom with the Play icon.

**Expected results**
A slider bar at the bottom with a Play icon.

**Project file**
N/A

**Log file**

**Additional information**
The Laptop has always been clean installed for each version of Linux Mint installed. 19.2, 19.3, 20
Tried Deleting .local/shares/cura but did not help.
I found out it should have a bar at the bottom from watching YouTube.

nallath, 2020-07-17:
So you either have it set to "force compatibility mode" in preferences, or your drivers doesn't support opengl 4.1

In both cases it falls back to the older way of layer rendering, which doesn't support the play functionality.

closed 2020-07-17 by nallath

bteamfox, 2020-07-17:
I typed in Terminal 
glxinfo | grep "OpenGL version"
got back: OpenGL version string: 2.1 Mesa 20.0.8
Well its a very old laptop, no opengl 4.1 for me, thanks for your quick response. :)