Ultimaker/Cura #8080: Import Profile from gcode not working -> "setting version" is not written to gcode-file

wolfgangmauer, 2020-07-18:
Instance container xxx is outdated. Its setting version is 0 but it should be 14.

**Application version**
4.6.2

**Platform**
Windows 10

**Printer**
Any

**Reproduction steps**
1. Slice and "Save to file"
2. Import Profiles and choose saved file.
3. Profile will not show up

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4940912/cura.log)

Ghostkeeper, 2020-07-21:
Yeah, I could reproduce this issue. The interface says it successfully imported the profile, but it doesn't show up. It gives these log entries:

```
2020-07-21 14:08:46,515 - DEBUG - [MainThread] cura.Settings.CuraContainerRegistry.importProfile [186]: Attempting to import profile /home/trin/Gedeeld/Projects/SlicerTestModels/test.gcode
2020-07-21 14:08:46,546 - WARNING - [MainThread] cura.Settings.CuraContainerRegistry.addContainer [66]: Instance container ultimaker3_test is outdated. Its setting version is 0 but it should be 15.
2020-07-21 14:08:46,550 - WARNING - [MainThread] cura.Settings.CuraContainerRegistry.addContainer [66]: Instance container ultimaker3_extruder_left_test is outdated. Its setting version is 0 but it should be 15.
2020-07-21 14:08:46,554 - WARNING - [MainThread] cura.Settings.CuraContainerRegistry.addContainer [66]: Instance container ultimaker3_extruder_right_test is outdated. Its setting version is 0 but it should be 15.
```

nallath, 2020-07-22:
It has been going wrong since 4.4. So that means that the importing can't be fixed for g-codes made with those versions of Cura (since the version upgrade can't be run on it).
I have fixed the writing for 4.7, so that's at least something

closed 2020-07-22 by nallath