Ultimaker/Cura #8083: Top of step between concentric cylinders not always filled correctly

Lalufu, 2020-07-18:
**Application version**
4.6.1

**Platform**
Windows 10 64bit

**Printer**
Ender 3 Pro

**Reproduction steps**
I have a model which is basically two concentric cylinders, on top of each other. The top cylinder has a smaller radius than the top one.
Depending on the exact difference in radius, the top of the larger cylinder is not always sliced correctly, and gaps can occur that allow "looking inside" the model.

**Screenshot(s)**
In both screenshots the model on the left has a radius for the upper cylinder that is 0.925mm smaller than the radius of the bottom cylinder.
The model on the right has a radius for the upper cylinder that is 1mm smaller than the radius of the lower cylinder.

This is the last layer that belongs to the lower, larger cylinder. The right model has been sliced rather differently, and has a proper brim, covering the gap. The left model does not.
[image]

This is the last layer of the lower, larger cylinder, and the first layer of the upper, smaller cylinder. On the left model the gap is clearly visible.
[image]

**Actual results**
Depending on the radius difference between the cylinders, a gap may appear between upper and lower cylinder.

**Expected results**
No gaps visible between the cylinders.

**Project file**
[CE3PRO_test-1.zip](https://github.com/Ultimaker/Cura/files/4941808/CE3PRO_test-1.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4941813/cura.log)

smartavionics, 2020-07-18:
You need to set the skin removal width to 0 to stop it removing that narrow skin region.

[image]

Lalufu, 2020-07-18:
Hi,

On Sat, Jul 18, 2020 at 07:55:43AM -0700, Mark Burton wrote:
> You need to set the skin removal width to 0 to stop it removing that
> narrow skin region.

that fixed it. I was pretty sure I was doing something stupid, but I
could not find a setting that seemed to fit the problem.

Thanks!

closed 2020-07-19 by Lalufu

Ghostkeeper, 2020-07-21:
This is in fact the [stereotypical example](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill/skin_preshrink.md) of what can go wrong if the setting is too high.

Lalufu, 2020-07-21:
I had not changed that setting before, so it was using the default value
(which is calculated based on other settings, as I understand). I have
not looked into what goes into that calculation, but the defaults seem
to be a bit optimistic.

However, slightly adjusting the value works fine for my case.