Ultimaker/Cura #8084: Cura print time accuracy with DIY printers

abdulmannan12, 2020-07-19:
Hello, I had build my own 3d printer with marlin firmware installed. I have been using cura for a while and encounter that the print time shown by the cura is way off than the actual printing time. I figure out that the reason is that acceleration and jerk speed set in the marlin firmware is not taken in account by cura software while calculating time estimation. I know that there is a acceleration enable feature in speed but I don't want to use cura to change the acceleration. 
I think that just by putting acceleration and jerk speed to printer setting to let cura know what our printer acceleration and jerk speed is to calculate the time only.

Thanks

fieldOfView, 2020-07-20:
If you install the "Printer Settings" plugin from the Marketplace, you can edit the acceleration and jerk settings Cura uses for its estimation in the print settings.

Ghostkeeper, 2020-07-21:
We're explicitly hiding these settings because we feel like people normally shouldn't change them, unless they build their own 3D printers. By hiding them, we're removing information that can be confusing and irrelevant to most people. This is good practice to make a UI that's easier to understand. Most people who build their own 3D printers will know how to edit a JSON file, or use the Printer Settings plug-in which also makes them visible again in the interface.

So we'll not make these visible. FieldOfView's plug-in makes for a good workaround. If you can read the speeds/accelerations/jerks from Marlin's configuration headers, you should be able to match them to the relevant settings in Cura, although Cura is slightly less powerful than Marlin in what is separately configurable nowadays.

radry, 2020-08-02:
Although with fieldOfView's plugin (" Printer Setting") the time estimation improves, it's still very inaccurate when using linear advance. Would it be possible to consider linear advance for the calculation of the print time?

I have also found fieldOfView's plugin "Linear Advance Settings" but that doesn't seem to be considered in the calculation and just adds a M900 command to the gcode.

nallath, 2020-08-03:
If the engine doesn't calculate it (which is the case for the Linear advance settings plugin), it's also unable to calculate the time. So as long as it's a separate plugin, the time estimation will be incorrect.

If you want better support for Linear advance, please make a separate ticket for that.

closed 2020-08-03 by nallath