Ultimaker/Cura #8085: Cura 4.6.2 crashes shortly after startup / program load

keyboardcowboyaz, 2020-07-20:
**Application version**
Cura 4.6.2
**Platform**
Ubuntu 18.04
**Printer**
Ender Pro 3

**Reproduction steps**
Tried Reinstalling Cura

**Screenshot(s)**
https://i.imgur.com/AWW3bxy.png

**Actual results**
Cura program load...then shortly after, the crash report is displayed

**Expected results**
Cura Starts normally

mahtDFR, 2020-07-20:
Looks like something to do with USB printing.
What happens if you remove your [user data](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations) (zip it up to make a backup first!)

Ghostkeeper, 2020-07-21:
I saw the crash report you had apparently sent with that crash dialogue, and fixed it here yesterday: https://github.com/Ultimaker/Cura/commit/8c4533a2c38cd9d390484aafa22b5533614d1130 ... The system works!

The crash report we got can be seen here: https://sentry.io/share/issue/02eb4dcc1c2646efb7d9d63ccefe5e41/

closed 2020-07-21 by Ghostkeeper