Ultimaker/Cura #8086: Missing warning when printing a material profile not compatible with printer

jorisa, 2020-07-20:
**Application version**
Cura 4.6

**Platform**
Windows 10

**Printer**
Ultimaker S3

**Reproduction steps**
Print using a material that has no compatibility profile for selected printer

**Screenshot(s)**
N/A

**Actual results**
Print is sent to printer, printer shows warning "change material" but the material is not available from the material list since it is not compatible with the printer.

**Expected results**
When clicking "Print over network" a warning should popup saying that the material profile is not compatible with the selected printer

**Project file**
N/A

**Log file**
N/A

**Additional information**
N/A

Ghostkeeper, 2020-07-21:
We do colour the material orange when you select it, and the header bar says "Not Supported".
[image]

Maybe it would help to make the profile drop-down orange as well? I'd expect you to see the orange colour when choosing the material in the drop-down. You wouldn't see it if you didn't select the material from the drop-down but by syncing with the printer, but then you'll have seen a warning on the printer about it.

fieldOfView, 2020-07-21:
My Sidebar GUI plugin implements a subtle but very useful warning icon to show which combination out of multiple extruders is causing the "not supported":
[image]
In the above we see that the combination in extruder 2 is the problem.

Ghostkeeper, 2020-07-21:
As @SandervanG mentioned in a duplicate of this issue, the UX designers have been trying to make this a bit more clear as well, but that's not finished. One related problem is that this can also be a problem of having the wrong nozzle. We'd show the warning on both the nozzle and material drop-downs:
[image]

This design also includes making the warning colour visible for a specific tab. The warning icon idea from fieldOfView is a bit more clear in that regard I think.

fieldOfView, 2020-07-21:
I could make a PR to get the icon in Cura.

fieldOfView, 2020-07-21:
The design posted by @Ghostkeeper above looks like it still needs the configurations floater to be opened for the problem-combination to be highlighted.