Ultimaker/Cura #8090: [4.6.1] Infill Layer Thickness != layer height increasing infill line width and removes extra infill wall count

whyme12, 2020-07-21:
**Application version**
Ultimaker Cura 4.6.1

**Platform**
Windows

**Printer**
Anycubic i3 mega, but also if I select a different model, like Ultimaker Original

**Reproduction steps**
1. Set Layer height to value X
2. set Infill Layer Thickness to value 2*X
3. Set extra infill wall count to some value > 0
3. Slice an object

**Screenshot(s)**
[image]
Left side: both values set to 0.1 mm, right side: infill line thickness set to 0.2 mm

**Actual results**
As seen in the screenshot, the left side is with both settings set to the same value, the infill width is shown correct in the preview and the extra walls are printed.
The right one on the other hand, the infill width is increased (except for the first 1-2 infill layers) and the extra walls are not printed.

**Expected results**
I would expect both sides to look the same, width he exception, that the infill on the right side is only printed on every second layer, due to the increased infill layer height

**Project file**
[cura_bug_profile.zip](https://github.com/Ultimaker/Cura/files/4953885/cura_bug_profile.zip)

**Log file**
none

**Additional information**
I was trying to increase my print speeds by setting 1 shell layer at low layer height and let the infill handle the other walls with increased layer height.

whyme12, 2020-07-21:
Additional information:
When printed, width of the infill is the same as the "infill line width" setting, but the extra infill walls aren't printed

whyme12, 2020-07-21:
retested it with 4.6.2, Bug still exists

Ghostkeeper, 2020-07-23:
I can indeed reproduce it in 4.6.2. In the current master though the entire Infill Layer Thickness doesn't seem to work any more which is a critical issue!

pdawwg08, 2020-12-06:
Still having this issue in 4.8, bug still exists. Also affects the support infill layer thickness in the same way. This creates weird infill, as the new width seems to interact with the other lines. As above this seems to delete some features or ignores density settings.

jellespijker, 2021-03-03:
This should be fixed for Arachne Beta when PR ultimaker/curaengine#1426 is merged. So hang in there guys. It won't be fixed in the next release with "old" engine. But it is coming very soon.

KimmoHop, 2022-03-10:
At least Arachne beta 2 didn't get the fix.
A bit hard to say if massively overextruded (note! I didn't test it beyond preview, so I'm not sure if it is just UI weirdness or does gcode have LxL extrusion when infill layer height is L times normal layer height?) infill in Cura 4.x is better or worse than missing infill in Arachne beta 2.

Ghostkeeper, 2022-03-20:
I see that it hasn't been mentioned before, but the infill lines appearing to be extra wide is expected behaviour (even in Cura 5). Cura's layer view can't visualise thicker layers for infill, and instead interprets the extra material as if the flow rate is set to 200%. It's not necessarily desired behaviour, but it's expected and accepted as a shortcoming.

The issue itself seems to have been fixed on Master at least. Looks good to me.