Ultimaker/Cura #8091: Lots of work being done even when not auto-slice, causes long delays

tlhintoq, 2020-07-21:
Easy to reproduce.
Put 40+ somethings on the bed.
Change any value in the settings.  Anything.  Wall thickness... whatever.

Even when "AutoSlice" is turned off there is a huge amount of calculations taking place that REALLY slow down Cura.  To the point where Windows may even report it not responding.  Eventually you can make one more change, and wait 2-3 minutes again.  One more change, wait 2 minutes.

And you don't dare kill Cura because you know that will corrupt all your profiles and machines and you'll have to nuke the installation back to default and set everything up all over again.

So you wait.

Can you just not calculate if the user has turned off "autoslice"?  I'd like to make my 20 changes, then I'll hit slice, then do all the calculations.  I know there's a ton of values based on values based on values... add a tick box for "pro mode" or something so its incombent on the user to know the values they are entering are decent.  But all the autocalculation to keep someone from hurting themselves is hurting my hourly productivity something horrible.

FYI: Quad core high speed machine with 24 megs of ram running no other programs - just no reason for it lag like this.

Ghostkeeper, 2020-07-23:
Cura has long had issues with having many objects on the bed. In 4.7 there are more improvements to this, but it remains an issue. I just tried your use case with 40 cubes, and it takes 2.6 seconds for me between the change of the setting and the first response from CuraEngine that it's starting the slice.

2.3 seconds of that is in the error check before the print. This is where we've made the biggest improvements, partially by preventing an error check if you don't have any per-object settings. If you'll post the log file, we'll be able to see if your problem is there. If it is in the error check, you'll probably have a similar effect as me in the next release and it'll be much faster. The 4.7 beta is not so far away any more either so you'll be able to test it for yourself too.

Do you have a log file for us?

nallath, 2020-07-23:
Well, it wasn't just the error checker, the rendering itself also had quite some unnecessary re-calculations going on. 

Also, I think I've managed to find the issue that caused the files to become corrupted. So that is also something that should be fixed for 4.7

closed 2020-08-06 by no-response[bot]