Ultimaker/Cura #8092: [4.6.2] Material configuration selection ignored

DecoyChild, 2020-07-21:
**Application version**
4.6.2
(The version of the application this issue occurs with.)

**Platform**
Windows 10, NIVIDIA GeForce GTX 1070
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
Ultimaker S5
(Which printer was selected in Cura?)

**Reproduction steps**
1. Removed Printer from network and reconnected.
2. Verified latest firmware in printer.
3. Attempted with model on build plate.
4. Attempted without model on build plate.

**Screenshot(s)**
Before showing Configurations available:
[image]

After selection, no changes were made: 
[image]
(Image showing the problem, perhaps before/after images.) 

**Actual results**
No Change
(What happens after the above steps have been followed.)

**Expected results**
Configurations should be selected/loaded
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
The configurations drop down clearly shows what is actually on the printer. Once it is selected, it does not change the slicing configuration. This is causing an error when trying to print, due to configurations not matching.
(Extra information relevant to the issue.)

Ghostkeeper, 2020-07-23:
Ha, nice. We had already seen this issue ourselves recently and already fixed this for the 4.7 version :grin: 

It's not expected to change the configurations per sé since the left extruder is the same in your case and the right extruder has no material. Cura doesn't allow having no material there, but it's expected to disable that extruder instead. So you'd have PLA0.4+PLA0.4, but the right extruder gets disabled. This is how it looks now in Cura 4.7 when you do this (I happened to have a printer with the exact same configuration as you).

[image]

Thanks for reporting anyway! Nice that you're paying attention and taking the time to report it :)

closed 2020-07-23 by Ghostkeeper

DecoyChild, 2020-07-23:
Excellent, thank you for the follow up. Disabling the second extruder was
my workaround so I could keep printing away. Glad to know it was fixed for
the 4.7 release.

It's rare I only have one material loaded, but I am using a lot of new
material and experiment a decent amount.

Again, thank you and keep up the great work 😁