Ultimaker/Cura #8093: 4.6.2. empty output files?

joshnosh, 2020-07-21:
**Application version**
4.6.2

**Platform**
win 10

**Printer**
flashforge creator

**Reproduction steps**
slice any file
load to SD card
try to print 

**Actual results**
nothing

**Expected results**
printing starts

**Log file**

[Files.zip](https://github.com/Ultimaker/Cura/files/4955250/Files.zip)

**Additional information**
output files seem to be empty

joshnosh, 2020-07-22:
reinstall attempted today. no change in result

mahtDFR, 2020-07-23:
Your log shows the slices are happening and it is writing files. Are you writing directly to the SD card? What happens if you save to your desktop?

Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so we can try to reproduce it.

Ghostkeeper, 2020-07-23:
You are facing a bug in the X3GWriter, a third-party plug-in maintained by me in my free time. This bug was already reported here, so please refer to X3GWriter's issue page: https://github.com/Ghostkeeper/X3GWriter/issues/30

joshnosh, 2020-07-23:
You are facing a bug in the X3GWriter, a third-party plug-in maintained by me in my free time. This bug was already reported here, so please refer to X3GWriter's issue page: Ghostkeeper/X3GWriter#30

found this out just befor i went to bed. uninstalled and reinstalled the plug in and it worked find

thanks for your app!

closed 2020-07-23 by joshnosh