Ultimaker/Cura #8094: [4.6.2] Linux Fedora AppImage doesn't open any webbrowser for login/signin

daton89, 2020-07-21:
**Application version**
4.6.2

**Platform**
Fedora 31 with KDE on a Dell XPS 9570 

**Printer**
Creality Ender 3 Pro

**Reproduction steps**
Clicking on Signin and then Signin or Create Account
Clicking Show online documentation from the Help menu. 

**Actual results**
Nothing happens

**Expected results**
 should open the browser to login.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4955972/cura.log)

**Additional information**
I checked some similar issues but found no luck. 
[#3072](https://github.com/Ultimaker/Cura/issues/3072)
[#3912](https://github.com/Ultimaker/Cura/issues/3912)

mahtDFR, 2020-07-22:
Your log shows this, might be related:

`2020-07-21 21:02:03,569 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._parseJWT [89]: There was no auth data or access token
2020-07-21 21:02:03,571 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._parseJWT [89]: There was no auth data or access token
2020-07-21 21:02:03,576 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._parseJWT [89]: There was no auth data or access token`

Are you able to try resetting your [user data](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations)?

mahtDFR, 2020-07-22:
This seems to work fine in master using Fedora

nallath, 2020-07-22:
It also works fine on the 4.6.2 app image on fedora for me (But I do have to note it's fedora 28).

daton89, 2020-07-22:
> Your log shows this, might be related:
>
> `2020-07-21 21:02:03,569 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._parseJWT [89]: There was no auth data or access token 2020-07-21 [...]

I just double checked and tried to remove the folders again: 

```sh
rm -rf ~/.local/share/cura
rm .config/cura/4.6/cura.cfg
```

and still not working

this is my updated cura log
[cura.log](https://github.com/Ultimaker/Cura/files/4960403/cura.log)

Ghostkeeper, 2020-07-23:
I'm thinking this is dependent on which browser is selected as the default browser. Can you try running this, see if it works?

    python3 -m webbrowser -t https://www.github.com

If there is no default browser, this would fail.

To be more alike the way that Cura opens your web browser, you'd have to execute this:

    python3 -m pip install PyQt5
    python3
    >>> from PyQt5.QtGui import QDesktopServices, QGuiApplication
    >>> from PyQt5.QtCore import QUrl
    >>> app = QGuiApplication([])
    >>> QDesktopServices.openUrl(QUrl("https://github.com"))

So if you have the will, you could also test that. We are so far unable to reproduce the issue though.

daton89, 2020-07-24:
Hi @Ghostkeeper thank you very much for your help 

I tried and my default browser (chromium) open github.com in both of your examples above

this is really strange and I don't know how to solve this problem

I just wanted to install some plugins but apparently I can't

Ghostkeeper, 2020-07-27:
Maybe it's a permissions issue somehow? I don't get how it works from a command line but not in our real code :(

Ghostkeeper, 2020-07-27:
Note that you don't need to be logged in to install plug-ins.

daton89, 2020-07-30:
How can I check permissions for the Appimage?

daton89, 2020-08-01:
The problem might be different, when I save a sliced file, the popup with `file saved, open folder button` appers, and when I click on open folder the folder nothing happens. 

Might be related?

Ghostkeeper, 2020-08-03:
I don't really know. Maybe something is up with Qt not being able to use your entire desktop environment.

Maybe this command is worth a try:

    env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP /path/to/Cura.AppImage

That seems to fix issues with the dark theme for some people, which is also something related to the desktop environment and Qt.

I notice that I'm automatically asking Linux users to do more difficult workarounds than other users. I guess I just think higher of them ;) Let me know if messing with terminals is difficult. But if this command works it could shed some light on what this problem is.

daton89, 2020-08-03:
You are great man! that solved both problems, the login etc buttons, and also the settings that were not visible because of the dark theme. 

I feel confortable with bash, dont' have to worry about that,  you are doing well. 

I'll make an alias in my `~/.zshrc` to open Cura with this command, but why do we need to unset those variables? 
Is there a way to fix permanently this problems?

many thanks for your support

Ghostkeeper, 2020-08-03:
It's hard to say. We haven't seen these issues in any of the developers' computers, even though 3 out of 5 work on Linux (which is way out of proportion for our userbase). That solution was found by [other users](https://github.com/Ultimaker/Cura/issues/5490#issuecomment-617252417) on a different issue.

darksaboteur, 2020-09-11:
I just experience this issue on Fedora 32 using the Cura 3.7.1 AppImage. 
The command Ghostkeeper provided fixed the issue