Ultimaker/Cura #8096: Please leave a built version of 15.04.6 to be downloadable

mmroden, 2020-07-22:
**Is your feature request related to a problem? Please describe.**
15.04.6 is the last version that works with New Matter printers.

**Describe the solution you'd like**
There used to be a build of this version available.  I would like a link to that version that does not require me to use a shady download site.

**Describe alternatives you've considered**
Using a shady download site.

**Affected users and/or printers**
New Matter printers.  The company is defunct, so I guess it's lower priority, but my god is it painful to use this thing now that this version of software is gone.

**Additional context**
(Add any other context or screenshots about the feature request here.)

mahtDFR, 2020-07-22:
It would be better that someone updates the definition to work with the latest version of Cura.
Documentation about how to do this exists here: https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura
The community is welcome to do such things, and we can review pull requests.

nallath, 2020-07-22:
I don't think I even have a 15.04.06 version anywhere anymore.

Ghostkeeper, 2020-07-23:
The oldest releases here on Github are there because I was secretly uploading them here without telling the web development team about it, as they wanted all downloads to go through the website. Eventually they had a website redesign and done away with all of their versions; now only these releases are left. It would truly be a shame if the old builds are gone. 15.04 wasn't even in this repository though, so no matter how diligently we've been tagging things we wouldn't have it here.

15.04 was particularly popular so you'll be able to find it online. This is probably the most reliable download: https://community.ultimaker.com/topic/29229-legacy-cura-version-150406/
I'm mirroring that file here: http://dulek.net/work/Ultimaker_Cura-15.04.06.exe

hkienle, 2023-11-21:
> I'm mirroring that file here: http://dulek.net/work/Ultimaker_Cura-15.04.06.exe
Now here it seems: http://dulek.net/work/ultimaker/Ultimaker_Cura-15.04.06.exe