Ultimaker/Cura #8098: Local Retry Obscures Print Information

akcooney, 2020-07-22:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Ultimaker 3

**Reproduction steps**
1. Start a print via Cura over the network.
2. Cancel the print locally at the Ultimaker 3.
3. Clear the aborted print.
4. The Ultimaker 3 will ask you if you want to retry.
5. Choose yes, and retry the print.
6. Attempt to monitor the printer via Cura or Cura Connect.
7. Note that print information is not available.

**Screenshot(s)**
[image]
[image]

**Actual results**
Print information is not available on either 

**Expected results**
I would expect print information for a retried print first issued over the network to be available in either Cura or Cura Connect.

**Project file**
Here for reference, but not sure if it's needed: 
[UM3_Standoff.zip](https://github.com/Ultimaker/Cura/files/4960397/UM3_Standoff.zip)

**Log file**
Log file attached: 
[cura.log](https://github.com/Ultimaker/Cura/files/4960404/cura.log)

**Additional information**
N/A

Ghostkeeper, 2020-07-23:
I can reproduce this indeed. The problem is in the firmware of the printer, not in Cura. I'll pass the bug report on, but no new firmware releases are planned for the UM3 any more unless there are critical issues with it. I don't expect that it will ever get fixed for this printer (even if there is a critical issue, they'd only fix the critical issue to reduce risk).

There's a chance that the same issue exists in the UMS5 though. I can't test that from home since I only have a UM3 here.

I'm going to give this report to the team responsible for printer management (which implements that page and also passes on the print information for Cura to display). They'll be able to test if it's still an issue in more recent releases and if so, fix it for those. I'm sorry, but I don't really have a solution for you, but your report will at least help others.

I'm closing this because it's not an issue in Cura but in Ultimaker's printers.

closed 2020-07-23 by Ghostkeeper

Ghostkeeper, 2020-07-23:
Apparently the firmware team already knew about this bug, and it indeed also happens for the S5. A fix might be published for the S5.