Ultimaker/Cura #8099: Add support for changing the filament between models

zouden, 2020-07-22:
I wish to be able to make part of a model a different colour (within the same layer), but I only have a single extruder.

**Describe the solution you'd like**
Ideally, we could select a material for each model (eg when models are overlapping, one can determine the colour used in the overlapping part). This is how it currently works for dual-extruder machines. But instead of choosing an *extruder per model* we should be able to choose a *material per model*. Cura would then insert an M600 filament change command at the appropriate part of the Gcode. It would also set the temperature and speed settings etc.

**Describe alternatives you've considered**
[This video from Make Anything](https://www.youtube.com/watch?v=KV2AjyowXX4) shows a way to do it with Cura but it requires doing multiple print jobs and enabling Z-hop. This is inconvenient and risky with printers that automatically level the bed before printing.  

A simpler alternative to what I proposed above is a post-processing script which inserts M600 at the appropriate part of the gcode, without providing a preview of the different colours in the main UI.

**Affected users and/or printers**
Users with single extruder machines that wish to use multiple colours in the same layer will benefit from this. The linked youtube video demonstrates where this is useful, ie in making phone cases with artistic patterns.

Ghostkeeper, 2020-07-23:
The video from Make Anything shows Simplify3D, not Cura. Maybe I missed something though (I skimmed over the video).

> A simpler alternative to what I proposed above is a post-processing script which inserts M600 at the appropriate part of the gcode, without providing a preview of the different colours in the main UI.

This is already in Cura! The "Filament Change" post-processing script indeed adds an `M600` command at a certain layer, and allows you to edit the parameters of that command (like how far it should retract the filament). It's more limited, since it only allows changing by layer. If you need it to change by extruder, I advise to use a Search and Replace script to replace the `T0` and `T1` commands with `M600`.

zouden, 2020-07-23:
The filament change plugin only works per-layer. This feature request is for per-model filament change.

>If you need it to change within a layer, I advise to use an actual multi-extrusion printer

Sure but in the absense of a multi-extruder printer, a single filament change within a layer allows for significant creative freedom.

Ghostkeeper, 2020-07-23:
What I mean is to use a multi-extruder printer in Cura, like Custom FFF Printer with multiple extruders, sorry.

I'd closed this because one of your alternative implementations was already implemented. I'll reopen this for your requirements, but we can't give this issue priority since it's not applicable to Ultimaker's printers and quite some work to make it work nicely. If we ever get a printer to support `M600` we can look into it again. In the meanwhile, this issue is a good starting point for an external contributor.

zouden, 2020-07-23:
Fair enough. I've tried the "custom FFF" printer method and it's rather tedious, because there's no way (that I could see) to copy all the settings that Cura uses for the Ender 3. I copied as many as I could but even then the quality was still worse. 

Ideally, we should be able to edit the printer profile for the Ender 3 and add a second extruder. Currently the number of extruders is hard-coded at 1, and when I modified the config file, I got a corrupted profile. Perhaps this should be the more immediate feature request.

DanHudson15, 2020-10-31:
I thought that the filament change inserted before a new layer, so the Z height would be correct upon resume.  I noticed tonight it is in the middle with no Z code near it.

;MESH:NONMESH
G0 F300 X258.972 Y238.83 Z0.84
G0 F9000 X262.665 Y238.83
G0 X262.653 Y56.847
G0 X40.9 Y56.835
G0 X40.9 Y60.4
;TIME_ELAPSED:25368.798475
M600 E30.00 L300.00 X0.00 Y0.00 ; Generated by FilamentChange plugin
;LAYER:6
;TYPE:WALL-INNER
;MESH:You are my sunshine(1).stl
G1 F1500 E290.85971
G1 X40.9 Y239.6 E294.79346
G1 X259.1 Y239.6 E299.58332

Ghostkeeper, 2020-11-03:
The Z command for the new layer is in the second line you posted. It moves to `Z0.84` in your case.

`M600` has an optional Z parameter to determine how much it should lift the nozzle during the pause. However this parameter is never output by the post-processing script. It should not change the Z coordinate for the pause.

If it did change the Z coordinate, I'm unsure what the spec says about whether it moves back automatically or whether that needs to be put in a separate command. But that's irrelevant now since we don't output any Z coordinate with the `M600` command.

closed 2023-07-13 by github-actions[bot]