Ultimaker/Cura #8100: [4.6.2] Scale XY mm and percent box is mixed

skjolddesign, 2020-07-23:
**Application version**
4.6.2 (same problem in 3.51)

**Platform**
Windows 10 x64

**Printer**
Ender 3

**Reproduction steps**
1. Unselect Snap Scaling and Uniform Scaling.
2. Reduce X mm by half.

**Screenshot(s)**
[https://prnt.sc/tndko3](https://prnt.sc/tndko3)

**Actual results**
Y percentage box is 50%.

**Expected results**
X percentage box should be 50%.

konskarm, 2020-07-27:
Hi @skjolddesign, I tried reproducing the issue in 4.6.1, 4.6.2, and latest master with no success. Halving the X mm corresponds to the X % being halved.

Ghostkeeper, 2020-07-27:
Did you rotate your object before trying this?

skjolddesign, 2020-07-28:
> 
> 
> Did you rotate your object before trying this?

Yes, I must haved done that. Only when model is rotated, is the scaling mixed.
I rotated the blue axis 90degr. with the clock.

nallath, 2020-07-29:
Duplicate of https://github.com/Ultimaker/Cura/issues/4136

closed 2020-07-29 by nallath