Ultimaker/Cura #8101: Cura 4.6.1 Windows application does not refresh print job monitor

rickmjohnston61, 2020-07-23:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
UM3

**Reproduction steps**
1. Submit print job status reported on monitor tab - OK
2. Submit new print job (without closing Cura) 

**Screenshot(s)**
Would not be helpful in any way

**Actual results**
First print job still shown printing despite being completed and new print job submitted

**Expected results**
New print job should be displayed

**Project file**
Would not be helpful

**Log file**
Would not be helpful

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-07-27:
I can't reproduce this problem. This is how the monitor page looks after submitting my first print job:
[image]

And this is how it looks after submitting the second print job:
[image]

It added the job to the queue, without restarting Cura anywhere along the way. Is that not happening for you?

rickmjohnston61, 2020-07-27:
Thanks for the reply but that is not the issue. Yes I can add multiple print jobs however while it lists in the que 'waiting for next available' when the job starts or I submit a job after the first 1 finishes the page does not refresh and still shows that previous job is still printing. I would take screen shots but I am currently no longer printing and do not have time to do so.

Thanks again.

Ghostkeeper, 2020-07-30:
It's expected that the print job stays active in the "finished" state if the print is finished but you haven't confirmed on the printer yet that the print is removed from the build plate.

Can you elaborate on the reproduction steps, maybe? With your latest information I'm currently trying this:
* Slice something.
* Press on "Print via Network".
* Press again, sending a second print to the printer.
* Switch to the Monitor page.
* Wait for the first print to finish. -- The monitor page then changes the progress bar from 100% to "finished".
* On the printer, remove the print and push the wheel to indicate that the print has been removed.
* Watch the monitor page again -- After a few seconds, it indicates that the next print is printing.

First print almost done:
[image]

First print is done. On the printer it displays that you must remove the print and press the button to confirm removal:
[image]

After confirming removal:
[image]

rickmjohnston61, 2020-07-30:
@Ghostkeeper Yes those are the steps I am doing as well. My monitor page never updates to the subsequent print job. I've left it there on the monitor page as well as switch back n forth to slice page or the preview and then back to monitor page...it never updates just continues to read the previous print job and 100%.
I know it is communicating because I have also reviewed on the Ultimaker mobile app and that is correct and updates after I click at the printer print removed.

Ghostkeeper, 2020-08-03:
Well, I'm not able to reproduce the problem here.

That app uses push notifications which is a bit different from the way in which Cura works: Cura will poll the printer every 2 seconds.

If you click on "Manage in browser" in the monitor page, do you see that it's updated there?

closed 2020-08-17 by no-response[bot]