Ultimaker/Cura #8102: [4.6.1] Additional Fan Speed Threshold Setting

CCS86, 2020-07-23:
Unless I am missing something, there seems to be a lack of control available for how the cooling fan transitions from "regular speed" to "maximum speed".

There is a "regular/maximum fan speed threshold" setting. But, according to the description, layers which are longer than this time are locked to "regular speed". Then, layers which are shorter than this time begin ramping towards "maximum speed". This leaves you no control over that ramp rate. 

All it would take is one more threshold setting to act as a layer time, which at or below this value, you get "maximum speed". That would give you control of the start and end layer times to define the ramp.

The current behavior seems to aggressively ramp towards maximum speed, once you cross the threshold. With [regular speed = 50%] and [maximum = 100%], this causes rapid changes in fan speed, which cause visible print artifacts.

Ghostkeeper, 2020-07-27:
It's a linear interpolation:

[image]

Between "Regular/Maximum Fan Speed Threshold" and "Minimum Layer Time", it should linearly interpolate between "Regular Fan Speed" and "Maximum Fan Speed".

However there is a known [bug](https://github.com/Ultimaker/Cura/issues/5409) with the fan speed interpolation right now. Something is going wrong in CuraEngine there.

Would you be okay with closing this as a duplicate of the buggy fan speed behaviour, or would you like to suggest a different behaviour than the linear interpolation?

CCS86, 2020-07-27:
Thank you. I didn't see anything in the minimum layer time about it also acting as the second fan speed threshold.

I think that decoupling [maximum fan speed] from [minimum layer time] would be an improvement. Since Cura is actively avoiding hitting the minimum layer time, by manipulating print speed, it's a bit of a "moving goal post". IMO, I would want to arrive at maximum fan speed before I ever start slowing down the print because of layer time. The primary goal being to maintain commanded speed, by using maximum part cooling. Then, as a backup resort, slowing things down.

Does that make sense?