Ultimaker/Cura #8103: There is no deltabot printer on cura 4.6.2？

ss8688, 2020-07-24:
I can‘t find deltabot on 4.6，but I seem to see it on 4.2

fieldOfView, 2020-07-24:
The use of the "deltabot" definition was deprecated over using the "Custom FFF Printer".

Use the Custom FFF Printer instead, and set the "Buildplate shape" to "Elliptical" and set the "Origin at Center" option. This is functionally equivalent to using the former deltabot definition, except you can now also choose multiple extruders.

Ghostkeeper, 2020-07-27:
The definition is still in our source code, so existing printers should remain functioning, but you can't add a new one. Maybe we need to transition people to the real Custom FFF Printer with a version upgrade...