Ultimaker/Cura #8104: [4.6.2] -  Nozzle Size von Nexus  wird nicht aus 4.6.1 übernommen  für Artillery Sidewinder X1

Kaicura, 2020-07-24:
**4.6.1**
(The version of the application this issue occurs with.)

**WIN 7 64 **
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Artillery  Sidewinder X1**
(Which printer was selected in Cura?)

**Reproduction steps**
Nach Installation  4.6.2 sind alle Profile in der  Version 4.6.1 weg.
Die Profile sind alle in 4.6.2 obwohl ich die 4.6.1 nicht deinstalliert habe
.
Den Drucker hat 4.6.2 auch übernommen aber nicht die Düsendurchmesser. Da ist jetzt nur noch die  0,4er drin.
Drucker und Nozzel config stammt von 3d nexus
es läst sich auch keine Düse von Hand einfach hinzufügen

Ghostkeeper, 2020-07-27:
Dit klopt, ja. Er is een versieupgrade geweest tussen 4.6.1 en 4.6.2 om een kritiek probleem met de maximale resolutie op te lossen. Omdat 4.6.1 en 4.6.2 dezelfde map gebruiken om hun configuratie in op te slaan, kan 4.6.1 niet meer de bestanden begrijpen die nu zijn geüpgrade naar 4.6.2. Dit is helaas een neveneffect van deze upgrade die we hebben moeten accepteren. Je kan feitelijk niet deze twee versies naast elkaar gebruiken. Sorry! We kunnen dit probleem niet oplossen (en zeker niet nu we 4.6.2 hebben gereleased).

Hopelijk hoeven we voorlopig niet meer nog zo'n versieupgrade te doen binnen een patch-versie. Er zijn best wat dingen mis gegaan met de 4.6 release daaromheen.

Ghostkeeper, 2020-07-27:
It would help to get bug reports in English next time. We don't speak every language in the world!

closed 2020-07-27 by Ghostkeeper

Kaicura, 2020-07-27:
Thanksfor regard