Ultimaker/Cura #8108: Slice when you press enter

zaquarius, 2020-07-25:
Automatic slicing is nice, so I don't have to click it every time, but then when I want to change multiple settings at once (especially complex settings that take a while to slice) I often have to wait for it to load. It would be nice if I could just press enter to slice (just make it an option under 'slice automatically'). Imo the only thing really missing from cura right now.

Thanks

chepo92, 2020-07-26:
I think I remember previous versions had a buton to pause auto slice next to slice button

fieldOfView, 2020-07-26:
That pause/resume button is now a plugin:
https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/PauseBackendPlugin

I think a hotkey to start slicing (when auto slicing is turned off) would be a good addition. Perhaps Ctrl+Enter.

chepo92, 2020-07-26:
> 
> 
> That pause/resume button is now a plugin:
> https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/PauseBackendPlugin
> 
> I think a hotkey to start slicing (when auto slicing is turned off) would be a good addition. Perhaps Ctrl+Enter.

That one was!, i knew i had saw in my cura at some point, after some reinstalls I forgot to add the plugin, thanks for it anf for octoprint plugin Aldo

Ghostkeeper, 2020-07-28:
There is a hotkey for this: Ctrl+P. Like, for printing.

nallath, 2020-07-28:
Well, since it's already there, I guess this can be closed.

closed 2020-07-28 by nallath