Ultimaker/Cura #8109: [4.6.1] Slicing skips a layer

louiswouters, 2020-07-25:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
1. Load the STL file in Cura
2. Slice the profile with default setting on Standard Quality (0.2mm)

**Screenshot(s)**
[image]

**Actual results**
one of the layers is skipped completely

**Expected results**
The print is a continuous part

**Project file**
[Layer Skip.zip](https://github.com/Ultimaker/Cura/files/4976738/Layer.Skip.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4976751/cura.log)

**Additional information**
When rotating the part around the z-axis, other layers are sometimes skipped. Changing the layer height also seems to have an effect on which layers are skipped

smartavionics, 2020-07-26:
The model isn't manifold (watertight) so you can expect there to be problems. Install the mesh tools plugin from the marketplace to get these warnings.

[image]

closed 2020-07-28 by nallath

louiswouters, 2020-07-28:
Thanks, I will install the plugin.