Ultimaker/Cura #8110: [4.6.2] Settings Dialog Doesn't Properly Inherit Desktop Theme

will-janz, 2020-07-25:
**Application version**
4.6.2 AppImage

**Platform**
Kubuntu 20.04

**Printer**
N/A

**Reproduction steps**
1. Run Cura AppImage in Kubuntu 20.04 with Breeze Dark theme
2. Open preferences 
3. Some text has extremely low contrast (materials section is especially egregious)

**Screenshot(s)**
[image]
(Using Ultimaker Dark theme)

**Actual results**
Certain sections are completely unreadable.

**Expected results**
Readable dark mode text.

**Project file**
N/A

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4976801/cura.log)

**Additional information**
Problem persists in Ultimaker Dark and Colorblind Assist Dark.

will-janz, 2020-07-25:
System dark theme with Ultimaker Dark:
[image]

System light theme with Ultimaker Dark:
[image]

It seems like some backgrounds aren't getting the correct color settings. I'm not familiar with this ecosystem, but if someone would like to point out what files to look at (I've poked around and I'm completely lost,) I'd like to help.

nallath, 2020-07-27:
These pages can be found here https://github.com/Ultimaker/Cura/tree/master/resources/qml/Preferences

Ghostkeeper, 2020-07-28:
Duplicate of https://github.com/Ultimaker/Cura/issues/5490.

See also the workarounds they posted there.

closed 2020-07-28 by Ghostkeeper