Ultimaker/Cura #8111: [4.6.2] Won't load properly

Derrick355, 2020-07-26:
**Application version**
4.6.2

**Platform**
Windows 10, RTX 2070, i5-9600k

**Printer**
CR-10S

**Reproduction steps**
1. Launch Cura

**Actual results**
Has the black box with logo and text like "Loading Machines"
A white box/program opens soon after the black box disappears.
White box disappears within 5 seconds

**Expected results**
Cura to open properly.

**Log file**
https://hastebin.com/caqofiqazo.sql

**Additional information**
Was using Cura 4.2, had this issue. After installed 4.6.2 im still having the same issue.

Derrick355, 2020-07-26:
Wow soon after posting I found the issue.

I installed GPU Tweak 2 a while ago with both working fine, but i _somehow_ enabled the overlay and it was causing Cura to crash.
Disabled the overlay as I don't care for it, and now Cura is working.

To disable the overlay:

1. Open GPU Tweak 2 from ROG
2. Open the Monitor
3. Go to OSD options
4. Disable OSD

I don't know if Ultimaker can do anything about it but hopefully this will help someone.

closed 2020-07-26 by Derrick355

Ghostkeeper, 2020-07-28:
It was crashing at the moment when Qt should be creating the main window. That's probably the white square you saw. I'm afraid that we have very little to go on and the problem seems to be more within the Qt framework than in Cura. Perhaps we're using some weird combination of shaders that makes it crash with that GPU Tweak overlay.