Ultimaker/Cura #8115: [FR] Advanced Mold Settings

chepo92, 2020-07-26:
**Is your feature request related to a problem? Please describe.**
Actual mold output is not useful sometimes as the faces need to be split to unmould 

**Describe the solution you'd like**
1. Option to output the mold polygon object
2. Option to directly output the polygon as a STL (this would be very useful as it cab be used in some other 3D soft)
2. Split the mold parameters (ej, given a vector/mesh, planes, etc)

**Describe alternatives you've considered**
I have been reading the cura code for making the mold and try to get this done by myself (at least solution 1), but with the lack of documentation is making it a bit slow, as far as I reviewed, when mold is enabled it executes getMinkowskiHull polygon from ConvexHullDecorator.py, which comes from [Uranium Polygon](https://github.com/Ultimaker/Uranium/blob/master/UM/Math/Polygon.py), which returns a Polygon, I [read](https://community.ultimaker.com/topic/22776-mold-feature/) that me mold is never converted to a model, but I imagine that at least the polygon can be ouput in some format (raw array?) and then somehow be converted to a STL file (e.g. with numpy-stl), and I'm left there, so I ask you people of knowledge 

I haven't found a similar simple and quick tool or software as this option to make a negative mold from an STL, I imagine it can be implemented in blender (besides the classic boolean operation to make a rectangular mold)), getting the STL of the mold enables to fast edit in other software and get it plane split (third feature request is to split directly from Cura)

**Affected users and/or printers**
I think many have been stuck with mold mode having no option to unmould complex figures, (and no workaround more than using the classic substract from polyhedron method

**Additional context**
Isn't there a development forum?

nallath, 2020-07-27:
Have you had a look at OpenSCAD? There are multiple scripts for it that enable mold generation.

https://www.thingiverse.com/thing:32657
https://www.thingiverse.com/thing:31581

chepo92, 2020-07-27:
Man! I knew openScad had a minkowskihull function but I didn't knew it could read STL (and I declared myself an user of OpenScad), dishonor my cow. 
I see those are from 2012, how didn't I found them before?!
Thank you very much!

nallath, 2020-07-28:
Glad to help :) I've been looking at doing someting like this myself for some time now. Never got to actually doing it.

closed 2020-07-28 by nallath

Ghostkeeper, 2020-07-28:
The mould is generated as 2D layers only. We don't actually do a 3D offset, so this feature is going to be hard to build (and arguably out of scope; Cura is not model editing software).