Ultimaker/Cura #8117: Feature Req: Manually Drop to Build Plate or override automatic

UnfoldAntwerp, 2020-07-28:
**Is your feature request related to a problem? Please describe.**
I find myself frequently switching on & off the 'Automatically drop models to the build plate' setting in preferences. Generally I prefer to have it set to ON but then I often find myself in a situation where I would like to be able to lift models 1 mm. For example if they have an irregular bottom and need support because I would like to have support under everything and not this tiny spot that touches the build plate and the rest sitting on support. In that case I need to temporarily disable the setting and lift al models up once they are positioned correctly.

**Describe the solution you'd like**
I believe the most elegant solution would be to have a 'Drop to build plate' button under the Move function that could be manually triggered when the general preference is set to off. Second, a way to override the automatic setting when it is set to ON, for example when you manually type a value in the Z axis in Move.

**Describe alternatives you've considered**
-'Select face to align to the build plate' could be suggested as an existing alternative but doesn't work nicely on irregular surfaces, also you need to orient the camera to the bottom view where the grey build plate obscures the view.

**Affected users and/or printers**
I believen this is useful for all printers and everyone who frequently alternates between geometric models and more organic models.

**Additional context**
(Add any other context or screenshots about the feature request here.)

Ghostkeeper, 2020-07-30:
We'd also have to make this preference per-model then. Currently when that preference is enabled, all models drop to the build plate, not just the one you moved just then. But if it's in the move tool you'd expect to be able to change it for only the selected models.

UnfoldAntwerp, 2020-07-30:
I guess you are right, the minimum behaviour I would expect is that if you manually tweak Z axis that it would actually do that. Now it feels like a bug, you type a value in that field and nothing happens, it stays at 0. Maybe some indicator like in settings, where you see a little warning or return arrow if you tweaked it manually and the model is not on the platform.

UnfoldAntwerp, 2020-07-30:
Klicking the return arrow drops the model back to the platform.

Liger0, 2020-09-04:
Making different generic settings being different between the moment you import the mesh and the later options like movements would be useful too.

fvrmr, 2022-02-01:
This is related to our CURA-8102 on our backlog.

Piezoid, 2022-08-31:
I made this:
[image]
I'm going to make a PR, if that's what you are looking for.

closed 2022-09-14 by nallath