Ultimaker/Cura #8119: support blocker issue

wizbrewery, 2020-07-28:
**Application version**
(The version of the application this issue occurs with.)
4.6.1
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
windows 10 64 bit latest updates
**Printer**
(Which printer was selected in Cura?)
Creatly Ender 3
**Reproduction steps**
1. Import stl file (reproducable 100% with even simple cube)
2. Select model by ctrl A
3. Click on support blocker tool
4. Click resizer tool

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
No support blocker is visible and the model remains active. 

**Expected results**
Little grey cube that can be resized as expected

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-07-28:
I think you are missing a step in your reproduction steps:

3a. Click on the model at the location where you want to place the support blocker.

wizbrewery, 2020-07-28:
eek,  you're right,  thank you 👍 

> I think you are missing a step in your reproduction steps:
> 
> 3a. Click on the model at the location where you want to place the support blocker.

wizbrewery, 2020-07-28:
thanks @fieldOfView ,  that's solved it

closed 2020-07-28 by wizbrewery