Ultimaker/Cura #8121: Tabs for print settings - idea

Peter2055, 2020-07-29:
This is not an issue. Well actually it is. Scrolling through the settings is enormously frustrating. Hunting back and forth etc for that setting you know is there but can't find at the time. Please can you consider making tabs for each heading? Either that or make a full page for the settings instead of them being inside a tiny window. Honestly there is no need to see the preview while trying to find a setting. Prusa Slicer has this right IMHO.

rburema, 2020-07-30:
> Scrolling through the settings is enormously frustrating. Hunting back and forth etc for that setting you know is there but can't find at the time.

I know it's not what you want, but this is why we have the search functionality, which, in the upcoming 4.7, will be:
 - faster
 - able to search for keywords not in the setting itself, but in the description as well

Personally speaking, I like to be able to see certain settings of different groups at the same time, and I'd wager I'm not the only one.

Peter2055, 2020-07-30:
Hello Remco,Thanks for your time in making your reply. I totally accept your point, but you kind of support my position with that. You'd like to see the settings together in one place. Me too. Have a full size window for the settings or at least give people a choice with different customisable views. If the window for the settings was expandable it would go some way to solving it. I have a whole window real estate but only less than a quarter holds the data I need to see/change. It's kind of claustrophobic. Like a house with a beautiful garden with only one tiny window to see it through. Normally people like to see the garden and make big windows.If you made the settings window to cover half the screen it would show twice the information in one place . The whole screen, four times more.  I think this closely fits your argument about having everything in one place. Best regardsPeter

Ghostkeeper, 2020-07-30:
If you're comparing it to Slic3r, you'll find that there it's also impossible to see the settings together in one place. For instance, you can't see the Infill Pattern and the Infill Speed at the same time, because they are in different tabs. In Cura you can, by making only the settings you're using often visible, or by using the search filter.

If I understand, basically what you're asking is two things:
* Allow only one setting category to be open at the same time, to prevent you having to scroll in most cases.
* Change the layout so that there is more space, e.g. place settings next to each other in some cases (like Z Seam Position X and Y) or to have multiple columns.

Nallath also brought up the argument that having tabs allows for better locational memory of where each setting is in the display, which is a good point. But maybe not enough to warrant this change by itself; you can also get locational memory in a list of items.

We're also getting complaints about people thinking the settings list is too wide though, because they are unable to see the scene any more without closing the settings list. I think the workflow of editing the settings while looking at the layer view is very common, so I would disagree that it's better to make the settings list take up the whole screen. As for allowing only one category to open at a time, I can see the value for that but I can also see it being very annoying to some people.

Alternative solutions would be to remove the setting visibility preset "All" which people often enable and then start complaining that their list of settings is too long. Maybe it's a good idea to rename the Basic/Advanced/Expert presets to something like Focused/Expanded/Extensive to signal that it's often better to not make all of the settings visible at the same time.

Peter2055, 2020-08-10:
I didn't reply to the earlier post at the time. I still feel that it would be good it the settings could reside also in a separate full page window, giving users a chance to select the flying window to edit them or a full page window that populates the flying window that presently exists but I accept that the idea might be time consuming/ resource intensive for a product that's distributed free. 

Going back the the existing layout, each classification  of the settings can be contracted to show the title only. This is a useful feature to help reduce the contents of the window to only those items needed but it is still a work around for a problem that really need not exist.

The flying window, if expanded to the maximum size on my 13 inch Macbook covers the "siice" button and covers the build plate, covers the save button etc.It needs to be continually moved from side to side to uncover the other controls. Perhaps if I were to use it on a different device with a larger screen real estate it would be better but the MacBook is, by and large my goto device. 

I believe it was anchored previously and I wondered at the time it was changed, why? Maybe you had users who requested it changed?

fieldOfView, 2020-08-10:
> I believe it was anchored previously

It was. You can install the Sidebar GUI plugin from the Marketplace if you want to restore that bit of sanity.

Ghostkeeper, 2020-08-11:
> Going back the the existing layout, each classification of the settings can be contracted to show the title only.

Indeed, you can contract them by clicking on the title. They start off collapsed.

> The flying window, if expanded to the maximum size on my 13 inch Macbook covers the "siice" button and covers the build plate, covers the save button etc.It needs to be continually moved from side [...]

Indeed, you can resize the settings list so much that it overlaps. I'd have liked to limit the size to something slightly smaller but that turns out to be quite difficult because the top bar and the action panel are on completely different sides of the element tree of the interface. Hopefully we can find a solution to that some day. Until then you'll have to manually make it small enough so that it doesn't overlap. Or accept that it overlaps a bit (which can also be useful to see more settings).

> I believe it was anchored previously and I wondered at the time it was changed, why? Maybe you had users who requested it changed?

This was the result of qualitative user testing. The UX designer team had invited a group of users over to use Cura and use their designs while they looked on. I wasn't involved, but from what I remember was told, is that it was found that the combined left-to-right and top-to-bottom visual path was confusing, and that people needed more vertical space for their settings (which is indeed something we were hearing a lot back then here on Github too). So a top bar was designed to consolidate the workflow within a stage from left-to-right only. The material and nozzle selection was moved to a separate part of that top bar to make more space. The only exception to that left-to-right workflow is the slice button, but that seems to not have been a problem, at least to those users.

FieldOfView's Sidebar GUI plug-in solves the vertical space issue while retaining more of the old design from version 15.06 through 3.6.

DreadKnight, 2020-09-26:
I like Cura's UI overall, but the right side bar is somewhat tedious stuff. To fix it, perhaps it should be approached like Blender (or even Photoshop?) does: show icon for specific category of icons on an outer bar and when clicked, show all the settings related to that. Blender will include search field as well soon, like Cura does. So it would be kinda neat to sort of make the UI/UX a bit more consistent across the two apps as they're both amazing and foss.

Ghostkeeper, 2020-09-29:
There is an icon for each specific tab:
[image]
Only it also includes the title, next to each of them. Maybe you meant that it's all collapsed off screen to the side then?

Blender is kind of infamous for having a bad UI :smile: Although that became a whole lot better with 2.80, and they also just have an enormous amount of features they need to support.

DreadKnight, 2020-09-29:
I know there are icons (it's obvious, duh), but the UI feels clunky and slow the way it's currently designed to function.
I'm talking blender 2.8+ UI style, so yeah, sort of outside as a panel functionality wise, always visible without scrolling.
Pre-2.8x series those were displayed horizontally by default (tabs); yeah, UI used to be pretty bad, but it's getting there.

konskarm, 2020-10-01:
The reason we have the settings panel this way and cannot switch to the Blender-like version (vertical tabs) is that we want our users to be able to view settings from multiple different categories at the same time. Using the settings visibility you can make your own customized panel so that you can have all the settings needed for that print. The [Settings Visibility Set Creator plugin](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/SettingVisibilityProfileCreator) can also help you manage different sets of visible settings. The blender style allows for only one category to be visible any time, which is not what we want in Cura.

DreadKnight, 2020-10-01:
@konskarm Indeed, thought about it more after I replied and the thing is that Cura has very low amount of settings for some of the categories. I think it would be best to be as it is atm, but with icons on a left sidebar could act as some kind of bookmarks that would scroll right away on the right side content to that category. It's how PDF viewers usually do it, left side index that's visible, right side scrollable content.

Nice to know that there's a plugin, but I usually don't like to fiddle too much with the UI and stray away from the defaults for various reasons.