Ultimaker/Cura #8122: Scaling Z axis changes Y axis percentage & more

GIJ0E, 2020-07-29:
**Application version**
4.6.2

**Platform**
Windows 10 Version 1903 Build 18362.959
Intel HD Graphics 520

**Printer**
Custom - MeCreator2

**Reproduction steps**
Selected the model, switched to the scaling tool, unchecked the Uniform Scaling checkbox, used the Z axis scaling coordinate on the model, changed scaling.
Noticed the Y axis percentage changes instead of the Z axis. Z axis mm changes appropriately and model scales appropriately.
Also...
Found that changing the Z axis mm changes the Y axis percentage.
Changing the Y axis percentage does not always correctly change the Y axis scaling. It also changes the Z axis percentage but does not scale the model's Z axis, looks like it multiplies it?
Reset also does not change the Y axis percentage back to 100%.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
[image]

rburema, 2020-07-30:
Have you rotated the model?

If so, probably yet another duplicate of #4136 , a longstanding issue that proved surprisingly tenacious.

closed 2020-07-30 by nallath

GIJ0E, 2020-07-30:
I searched for an existing issue but failed to find #4136 
You're correct, it was rotated. My apologies for missing that detail.
Thanks!