Ultimaker/Cura #8123: Save to Removable Drive truncates filename at last period in project name

GIJ0E, 2020-07-29:
**Application version**
4.6.2

**Platform**
Windows 10 Version 1903 Build 18362.959
Intel HD Graphics 520

**Printer**
Custom - MeCreator2

**Problem**
When saving to removable disk, the filename is truncated if a period exists in the project name. If multiple periods exist, it truncates at the last period. 
For example, a project name of TestPartV1.2.3.4.5.6 would be truncated to TestPartV1.2.3.4.5.gcode and TestPartV2.1 would be truncated to TestPartV2.gcode 

It does not do this using Save to File button, only the Save to Removable Drive button.

I also believe it doesn't ask to overwrite an existing file when saving to Removable Drive.

rburema, 2020-07-30:
I can confirm this, I'll put it up for discussion. _Probably_ an easy fix.

> I also believe it doesn't ask to overwrite an existing file when saving to Removable Drive.

This last one is intended, as the 'save to removable drive' is supposed to be a quick option. If you desire to be guarded against overwrites, then use the normal save to file function, but then put it on your removable drive.

rburema, 2020-07-30:
Pushed a fix, will probably be in 4.7

closed 2020-07-30 by rburema

GIJ0E, 2020-07-30:
Thank you very much!

commit 1d55befb (2020-07-30): Fix missing part of filename w. dots for removable dr.

Project names are no longer stored with their extension internally, so the removal of an extra extension wasn't necesary. Fixes #8123