Ultimaker/Cura #8124: Creality Ender Pro 3 print settings issue

YEETmaxter, 2020-07-29:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-07-30:
Make sure 'origin at center' is unchecked. It should be by default, but I can find some reports online of people who struggled with that.

Otherwise, as far as we know, the offsets in Cura are 'correct', in as far as they where contributed by the community.

This might not be a problem within Cura, but with the (calibration of) the printer?

YEETmaxter, 2020-07-30:
Thank you, rburema. Origin at center was not checked. Just want to make sure all our offsets (especially G-codes) are correct. Might you or anyone know anywhere we could find all correct coordinates? I have tried contacting Creality - no response as of yet. Sigh.

rburema, 2020-07-30:
Though I seem to remember this coming up in a different context. It might depend on the firmware? In that case, you could perhaps try x/y = 220x220mm since that is the case for the not-pro version.

YEETmaxter, 2020-07-30:
I'll look into firmware - many thanks, rburema.

Ghostkeeper, 2020-08-03:
I've also seen that the end stops are sometimes mounted wrong or inaccurately on the Ender 3 Pro which would make it print off-centre.

Ghostkeeper, 2020-08-03:
We are not in a position to really help you with this, because we don't have an Ender 3 Pro in the Ultimaker office. If you've found what the problem is, can you let us know?

closed 2020-08-17 by no-response[bot]