Ultimaker/Cura #8125: [4.6.1] Infill mesh gets printed when overlapping cutting mesh

Block137, 2020-07-30:
**Application version**
4.6.1

**Platform**
Win10

**Reproduction steps**
1. Add a normal model.
2. Add modifier mesh as "Infill mesh only"
3. Add another modifier mesh as "Cutting mesh"
4. Make the 2 modifier meshes overlap (does not need to overlap with model from step 1)
5. Slice

**Screenshot(s)**
Coming soon

**Actual results**
Overlapping region of the 2 modifier meshes gets printed even though it is outside of actual model.

**Expected results**
Modifier mesh region that does not overlap with actual model won't be printed.

**Project file**
Coming soon

**Log file**
None provided

**Additional information**
Infill mesh order has no effect.

nallath, 2020-08-05:
We've added this to our backlog: CURA-7636

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

Block137, 2023-06-11:
Still happening in 5.3.1
[image]
[image]

MariMakes, 2023-06-27:
Hey @Block137,

👋 Fresh eyes reporting for duty. 
Thanks for double-checking this behavior in 5.3.1 and letting us know. 💪 

Can you help me understand, how this bug is limiting how you are working with Cura? 
Do you have an example of a time when you had overlapping meshes that were printed when you did not want them to?

This extra context makes it easier for us to understand the priority of this bug. 
Thank you 🙏

closed 2023-07-11 by github-actions[bot]