Ultimaker/Cura #8128: Can't slice anything with version 4.6.2 | 4.6.1 | 4.6

Digital1O1, 2020-07-31:
**Application version**
(The version of the application this issue occurs with.)
4.6.2, 4.6.1, 4.6

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

OS : Windows 10
Laptop : Aorus 7 Gaming laptop
GPU : Nvidia GTX 1660 Ti

**Printer**
(Which printer was selected in Cura?)
Ender 3

**Reproduction steps** 

I can’t slice anything no matter which version of Cura, nor the STL file I use.

1) Took various STL files from Thingiverse to check if it was just the STL file itself, but I still couldn’t slice anything.
2) Restarting my laptop didn’t help.
3) Nor did deleting any Cura related folder on my Windows system before uninstalling/reinstalling Cura.
4) Doing an entire factory reset also did nothing.
5) Double checked the windows firewall to see if Cura was 'allowed' and that checked out fine.
6) Switching between both the integrated graphics card and the GTX  didn’t help
7) Running Cura as an admin didn't help.
8) When I checked the task manager the ‘Slicing engine for Cura’ wasn't doing anything and neither my CPU, integrated graphics card, or GPU was under any sort of load.
9) Switching between the default Cura profiles and the ones I’ve either downloaded or changed myself didn’t help
10) I double checked if both my graphic drivers were up to date and they were

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

[image]

**Actual results**
(What happens after the above steps have been followed.)

I can’t slice anything. 

The slicing progress bar will move a little bit, then will revert back and nothing will happen no matter how long I wait.

**Expected results**
(What should happen after the above steps have been followed.)

….I should be able to slice something

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

Here's is the wetransfer.com link to the Yoda project : https://we.tl/t-K7rprhQmK5

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues
 to find the log file to upload, or copy a relevant snippet from it.)

Here's the latest log file I have as of 7/30 at 7:43pm

[cura.log](https://github.com/Ultimaker/Cura/files/5004410/cura.log)

**Additional information**

(Extra information relevant to the issue.)

closed 2020-08-02 by Digital1O1

fieldOfView, 2020-08-02:
I don't know if you meant to open, close, open and close the issue again, but this looks like a duplicate of https://github.com/Ultimaker/Cura/issues/8136

Digital1O1, 2020-08-02:
> I don't know if you meant to open, close, open and close the issue again, but this looks like a duplicate of #8136

That's correct. 

I'm currently working on a project that heavily relied on 3d-printing and I gotten so used to Cura and needed/was hoping for  a quick response.

Hence the two tickets.

And I'm pretty new to reporting bugs to Github, so I assumed that closing the issue would 'close it out'