Ultimaker/Cura #8129: slicing issues with version 4.6.2

compudude1991, 2020-07-31:
**Application version**
4.6.2

**Platform**
Windows 10 home, intel core I5-10400F, NVIDIA GeForce GTX 1660 SUPER

**Printer**
MP select mini V1

**Reproduction steps**
tried several layer heights, wall thicknesses, and infill densities.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Expected results**
downloaded the same STL from onshape to an older machine running version 4.5.0 and the model sliced perfictly

**Project file**
[project.zip](https://github.com/Ultimaker/Cura/files/5005020/project.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5005026/cura.log)

**Additional information**
this was a fresh install, i havent done so in quite a while, i appologize if this is simply a setting issue that i was overlooking.

smartavionics, 2020-07-31:
Hi, the file you provided isn't a project file, it simply contains the model (STL). A Project file is created by doing File->Save in Cura, you would need to zip or rename the .3mf file to attach to this issue. However, I sliced your model and it looks OK.

[image]

Your image looks like you have coasting enabled which does cause gaps like that to be displayed.

compudude1991, 2020-08-02:
I just checked and it turns out that was the case. 
since it wasn't technically a bug, just an experimental feature that I wasn't aware of, I apologize for the false bug report. 
thank you for your assistance.

closed 2020-08-02 by compudude1991

smartavionics, 2020-08-02:
No need to apologize, cheers!