Ultimaker/Cura #8131: Speed up the loading of Cura

meltonpieman, 2020-07-31:
**Is your feature request related to a problem? Please describe.**
The one feature that makes me opt for Prusaslicer (not such a good product) as a "go to _first_ option" is the time that Cura takes to load on my laptop (even thought it is an i5 7th gen with 24gb memory). Of this time, the longest identifyable part is the time for 'Loading Machines'. I have only 1 machine, so the countless others are wasted time for me.

**Describe the solution you'd like**
The ability in setting up Cura to limit the machines it loads to those that the users can choose from the exhautive list available.

**Describe alternatives you've considered**
Nil

**Affected users and/or printers**
All users who do not have Cura running 24/7 will save wasted loading time.

**Additional context**
Nil

fieldOfView, 2020-07-31:
You can reduce the "Loading machines" stage by installing the "Startup Optmiser" plugin from the Marketplace, and using the `Extensions -> Startup Optimiser -> Disable loading unused configuration files` menu.

nallath, 2020-07-31:
I've fixed a lot speed / performance bottlenecks for Cura for 4.7. The loading of all the profiles should be a lot faster already.

closed 2020-07-31 by nallath