Ultimaker/Cura #8133: [4.6.2] broken supports

Liger0, 2020-08-01:
**Application version**
4.6.2

**Platform**
Windows 10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Enable supports

**Screenshot(s)**
[image]

**Actual results**
1. The supports in the inner part of the model have some missing layers
2. The supports in the inner part of the model don't have a brim
3. There is support brim on the outside even though there is no support

**Expected results**
None of the 3 above issues

**Project file**
[broken supports.zip](https://github.com/Ultimaker/Cura/files/5009419/broken.supports.zip)

Ghostkeeper, 2020-08-04:
It would maybe help to have issues in separate threads if you'd like the labels to be anywhere near correct. But here goes.

1. This is an issue with the Maximum Resolution algorithm that is fixed in Cura 4.7.
2. They do, but the support brim goes towards the inside, not the outside. Seems like there is a bug that Support Only on Outside doesn't recognise that there are pieces inside the hole here because those pieces are support rather than mesh. I see that as a bug and see what we can do to fix it.
3. The fix for this is Minimum Support Area, which you've overridden to 0mm². Setting that to a more reasonable 1mm² helps. Is there a better technique you can think of to strip pieces of support that are too small to print?

Liger0, 2020-08-04:
For the third issue, maybe checking if there is any support or model mesh the second layer, and if there isn't avoid the brim generation would be useful. 
This would require a bit of prediction tho and, considering the support implementation makes the support already start on first layer and have a brim inside, rather than making a 1 layer "raft" (brim inside+outside including the intersection) and the support starting at second layer, I don't know if it's possible

Ghostkeeper, 2020-08-05:
There is also support in the second layer, but it's also too thin to print there. We can't easily see if there are any support infill lines at the stage where the brim is being generated since it hasn't drawn the support lines at that point yet.

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue. 

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-06-13 by github-actions[bot]