Ultimaker/Cura #8135: Tooltip Details for Hole/Horizontal Expansion

thescreensavers, 2020-08-01:
The current tool tip leaves out vital information that is mentioned in this documentation 
https://support.ultimaker.com/hc/en-us/articles/360012512340-Shell-settings

Current Tool Tips:
[image]

Id suggest to include  "Only affects X/Y Layers" somewhere in both tool tips as this has caused me a bit of headache with holes in the Z axis.

fieldOfView, 2020-08-01:
Doesn't the word "horizontal" imply the x/y plane already?

smartavionics, 2020-08-01:
Also, there are no "Z layers".

thescreensavers, 2020-08-01:
> Doesn't the word "horizontal" imply the x/y plane already?

hindsight is 20/20, adding clarity is helpful.

Ghostkeeper, 2020-08-04:
It used to be called X/Y Offset but that was deemed too technical.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant