Ultimaker/Cura #8137: Ender 5 Pro

jtoomb79, 2020-08-02:
There is no preset printer selection for the Ender 5 Pro. Is there another Creality printer I should use, or use custom?

NetTecture, 2020-08-02:
Ah, there is zero sense in having a Profile for the Ender 5 Pro - it basically is an Ender 5 with 3 changes:

* The silent board. It is there to make the printer SILENT - it has ZERO other functionality. It is sold basically to make the stepper motors silent.
* A metal extruder instead of the plastic one. The reason is that the plastic one is problematic, after some time (deformation), so they made a metal one.
* A capricorn tube to make the extrusion more exact. Note that Linear Advance (which would change settings here) is not used in the firmware.

ALL 3 changes are also sold separately. In fact, I personally consider the pro quite a useless upgrade - because the silent board is better replaced with a SKR E3 Mini 2.0 which is ALSO way faster, has more memory (for a custom firmware) AND even better drivers for a lower price.

Anyhow, as all those 3 changes are firmware and feature neutral in themselves, having an Ender 5 Pro profile makes no sense. It basically is a "Pro package" that does not change anything (as far as Cura is concerned). Having an Ender 5 profile DOES make sense. I think there is one ;)

jtoomb79, 2020-08-02:
**NetTecture, thank you for taking the time to reply**, it really helps someone printing their 1st benchy. Sorry if my post here is in the wrong format, I'm still on the steps of the shallow end of the pool (so to speak). 

That's what I was thinking, so I went with the Ender 5 profile (though, I was a bit confused since Cura has both Ender 3 and Ender 3 Pro profiles). I'm getting stringing and under extrusion, so I'll work with the Ender 5 profile to try and those sorted. 

I did plan on upgrading to the Mini E3, but thought I should learn the basics first. Would upgrading to the Mini help, or would I have the same issues? 

Cheers.

NetTecture, 2020-08-02:
Obviously by logic you would have the same issues with an E3. Here is why:

Stringing is bascially a function of temperature and other printer parameters - mostly retraction. It also depends on material, but optimization bascially runs along temperature, retraction settings and  travel speed (sometimes moving faster can make stringins less of an issue).

Replacing the motherboard does NOTHING to change any of those parameters. Actually even new firmware does little - mostly because when you use Cura, Cura sets those parameters via the settings you choose. The firmware defines maximums (max movement speed), but Cura defines what is actually used.

What you ask is similar to "if I install a navigation system in my car, will it stop me from getting speeding tickets when I go to fast" - totally unrelated. The E3 is IMHO vastly superior to the board you got on the Pro 5 - in fact, I consider now (after buying it) the whole Pro wasted money... rather get normal, E3 board and install a much better (and expensive) direct drive setup. But this vastly superior board means:
* More speed
* Better Drivers
* More memory for more and better firmware (the 128k on the pro board are extremely limiting once you start activating more features in Marlin - you bascially can not, out of memory). This gets interesting if you want to use BlTouch and Linear Advance AND SD card. Not possible on the old motherboard.

But srtringing will not magically change. You need to work on your parameters.

What I found out - based on MY filament - is that I get a LOT less stringing (nearly none) when I print:
* Higher temperature (though I am actively trying to see how fast I can get it). Talking of normal 210, up to 230 degree. I also am working on faster printing.
* MUCH faster retraction speed in the firmware
* MUCH larger retraction. 6mm+

That is for PLA. Oh, and the one that I got with the printer was utter garbage. Supposedly the quality of that roll varies. Once I used it up, I put in a Creality Filament and WHOW - it just worked. Some other people say their roll was good, so this initial roll sems t obe al ottery ticket. I would suggest having other filament ready. Mine also clogged 4 nozzles or so - ever since switching and runing my extruder pressure (I was like pressing it way too hard) I had ZERO issues. Obviously the extruder STILL is sub-par compared to a BondTech or clone - but the clones there go for 60 or so USD, the orirginal for more than 100.

I end up with a model and 4-5 very very very thin strings most of the time, though the head is also very bad in part cooling. At the end, you get what you paid for. NOT a criticism. It is a nice platform ,but it is not a high end printer. It is extremely cost optimized. But the, you are sold it as a lower budget printer, not a high end machine. You can get printers with 10 times the speed these days quite cheap - but this STILL is many times the price.

Under Extrusion: Ah ,calibrate the extruder. And check your Z-Axis (should be 800 steps/mm, my firware had 400, as do many others - which means loss of 50% of height movements, crazy. Anyhow, somehow the firmware has the extrusion nubmers wrong, which is funny given that the extruder steps/mm is a maethematical function depending on the gears, so it is not exactly "variable". It seems to be too low - which means under extrusion.

closed 2020-08-03 by nallath