Ultimaker/Cura #8138: Cura makes a bunch of unnecessary moves near end of layer.

Anon0ne, 2020-08-02:
**Application version**
(The version of the application this issue occurs with.)
4.6.2

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows x64
**Printer**
(Which printer was selected in Cura?)
Ender 5
**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

Slice, and print. Or check gcode you will see many G0 near end of layers.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

https://www.reddit.com/r/ender5/comments/i1vkuv/cura_46_making_a_bunch_of_unnecessary_moves/?utm_medium=android_app&utm_source=share

**Actual results**
(What happens after the above steps have been followed.)

Too many G0 near end of layers

**Expected results**
(What should happen after the above steps have been followed.)

No unnecessary time spent on bad moves.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

https://www.reddit.com/r/ender5/comments/i1vkuv/cura_46_making_a_bunch_of_unnecessary_moves/?utm_medium=android_app&utm_source=share

**Additional information**
(Extra information relevant to the issue.)

NetTecture, 2020-08-02:
Funny, I see this one too. In my case the printer goes to a specific place on the model in some slices and moves the extruder left and right.

And no, it is not extruding anything - because some of the models I print do not have a flat top and in some cases it is happily "wiping" over empty air. After a couple of cm (or between two models) it is obvious that this is not "filling small holes" - in my case it is basically air wiping.

smartavionics, 2020-08-02:
Please provide a project file that exhibits this issue. Do File -> Save and then rename or zip the .3mf file so you an add it to this issue. Thanks.

Anon0ne, 2020-08-02:
smartavionics, I believe it's pretty much every model that does it. I see multiple G0's happen near the end of most layers.

`;MESH:NONMESH
G0 F300 X188.324 Y52.095 Z10.3
G0 F6000 X188.271 Y52.235
G0 X188.233 Y52.283
G0 X187.222 Y52.293
G0 X143.868 Y52.44
G0 X142.868 Y52.284
G0 X142.954 Y52.389`

[CE3_Wemos.zip](https://github.com/Ultimaker/Cura/files/5013192/CE3_Wemos.zip)

smartavionics, 2020-08-03:
That's not a project file, it only contains the models and not the settings. As requested above, please do File -> Save (not export).

Anon0ne, 2020-08-03:
Ooops, sorry. I will fix that. Anyone could slice a file and see these
problems at the end.

smartavionics, 2020-08-03:
> Anyone could slice a file and see these problems at the end.

Sure, but I need to see the settings that are being used to understand what is causing the travels.

Ghostkeeper, 2020-08-04:
Most likely it's preparing to move to the start location of the next layer.

NetTecture, 2020-08-04:
No, it is not. There is zero need to move back and forth multiple times to "prepare" a move. What would "prepare a move" even mean? it is aslicer. it has GCode commands. THe only "prepare" that MAY be needed (and could this be related to this?) is a minimum time per layer. Can it be that it is moving around so the nozzle does not ooze (in one place)? That is about the only "prepare" I could logically think of.

smartavionics, 2020-08-04:
Until someone provides a project file that shows the settings in use, we can't do much.

smartavionics, 2020-08-04:
> Most likely it's preparing to move to the start location of the next layer.

I have wondered whether we should make the final travel move direct rather than combed when retraction is enabled given that the z-change happens before the travel.

Ghostkeeper, 2020-08-04:
> No, it is not. There is zero need to move back and forth multiple times to "prepare" a move. What would "prepare a move" even mean? it is aslicer. it has GCode commands. THe only "prepare" that MAY [...]

Sorry, I should just say "it's moving to the start position of the next layer" then.
I was thinking from a code perspective where this move is calculated separately from a different thread that assumes it starts in a certain position.

NetTecture, 2020-08-04:
Here we go. This is a Bondtech small fixture. See the same thingon the generated code - after every layer it is "swiping" (left and right multiple times) WHILE retracted before starting the next layer.

[Fixture.zip](https://github.com/Ultimaker/Cura/files/5022960/Fixture.zip)

smartavionics, 2020-08-04:
Thanks for the file. I tweaked a few settings and it much reduced the travel:

combing mode = not in skin
max comb dist with no retract = 10
avoid printed parts when travelling = disabled

[image]

Anon0ne, 2020-08-04:
> Thanks for the file. I tweaked a few settings and it much reduced the travel:
> 
> combing mode = not in skin
> max comb dist with no retract = 10
> avoid printed parts when travelling = disabled
> 
> [image]

But should you have to change all that? Shouldn't avoid printed parts when traveling make it not act like it's bouncing all over the parts before the next layer?

I just see it as a wasted time bug.. and I think even with combing and avoid travel it shouldn't do all the tiny circles and jittering.

smartavionics, 2020-08-04:
>But should you have to change all that? Shouldn't avoid printed parts when traveling make it not act like it's bouncing all over the parts before the next layer?

> I just see it as a wasted time bug.. and I think even with combing and avoid travel it shouldn't do all the tiny circles and jittering.

OK, I'm missing something here, please provide an image that shows the tiny circles and jittering etc so I can understand what you are talking about. Thanks.

Anon0ne, 2020-08-04:
I will get some more examples tonight, I lost power in Hurricane Isaias last night and couldn't do much from home..

What I mean is I have noticed at the end of a skin, or top/bottom mostly it will get to the corner area where there is a tiny amount to be done, and finish that. Then it begins to shift around the print into small corners, does a small dot or circle and then jitters around to a few other small areas and then continues on to the next layer. I have to find the proper models to show this well; but I will slice some in my downloads folder to send over.

Thanks for looking into it; and I am still unsure why you had to change all of that to reduce the traveling.. I never had this problem until late.

NetTecture, 2020-08-04:
I have another example here that shows the SAME behavior - it is the "old" 2x2cm test cube larger in 8x8cm and with "insance" speed settings (I acutally us that project to test extrusion limits). I can upload that tomorrow AND put a video on youtube.

There it basically "combs" on a side layer that is IN THE WALL - not on top of something but bascially it finishes a layer then does some cm back wnd forth rocking. on one of the walls, then goes to the next layer. This is not what ANY of those settigns SHOULD change. Let me put up a video ;(

smartavionics, 2020-08-04:
I won't watch the video but a a gcode or project file would be useful.

Anon0ne, 2020-08-05:
Here's a file making alot of random moves even where stuff isn't printed yet. Rename it to 3mf
[CE5_Listr_v1.1.zip](https://github.com/Ultimaker/Cura/files/5025910/CE5_Listr_v1.1.zip)

smartavionics, 2020-08-05:
Thanks for the file. Yes, at the beginning of the first layer it does some wiggly travel. This is due to the no skin combing mode which tries to route travel moves along the top of walls and infill rather than skin. So what it's doing there is following the walls to get to the first location where it is to print. I agree that isn't ideal but it's at least doing what it should (avoiding skin areas). Setting the combing mode to all will stop it doing that but then it may travel across skin.

Combing mode not in skin...

[image]

Combing mode all ...

[image]

Ghostkeeper, 2020-08-05:
> Here we go. This is a Bondtech small fixture. See the same thingon the generated code - after every layer it is "swiping" (left and right multiple times) WHILE retracted before starting the next layer.
> 
> [Fixture.zip](https://github.com/Ultimaker/Cura/files/5022960/Fixture.zip)

Thanks for the project file. Please disable the "Wipe Nozzle Between Layers" setting to prevent it from swiping before starting the next layer.

Ghostkeeper, 2020-08-05:
The combing moves posted by Anon0ne are indeed to avoid the skin, as indicated by the Combing Mode setting "Not in Skin", like Smartavionics said. This is the default for Creality Ender 5. In most Ultimaker printers it's set to "All" instead so that it doesn't need to retract as often which is more of a problem due to their Bowden tube.

But indeed myself I prefer "Not in Skin" since PLA retracts fine, and I don't care much about the time saved by cutting off those small travel moves.

The default has been set by people who made those profiles through extensive testing. We don't have that printer, so we won't make any modifications to it without getting a reasonable argument and tests to show that it's better.

Anon0ne, 2020-08-05:
That was only one example.. you can close this but that isn't the original problem. It's definitely a bug with 5-10 G0 moves at the end of a layer

Anon0ne, 2020-08-05:
[CE5_Hero_Me_Gen5_Base_5.zip](https://github.com/Ultimaker/Cura/files/5029692/CE5_Hero_Me_Gen5_Base_5.zip)
Rename this to 3mf.. and look at the end of the first layer. I set Combing to all to show you it's still doing the same thing, moving all around near the end of layers.

smartavionics, 2020-08-05:
That isn't a project file, it doesn't contain any settings. Do File->Save (not export!!!!!!)

Anon0ne, 2020-08-05:
I re-saved the project and added to previous post. If you guys say this is feature instead of bug I believe you as I am not a contributor nor have I looked into the code. Just had several people agree they noticed this happening now, and wanted to bring it up.

Thanks so much for your time and contributions!

smartavionics, 2020-08-05:
Turn off avoid printed parts when travelling and it will do less travel at the end of the layer.

Ghostkeeper, 2020-08-05:
It is a project file, or at least it is now. I think Anon0ne edited a better link in later.

The travel move Anon0ne is talking about is this one, at the end of the first layer (in black):
[image]

That is the shortest route to travel from the last gap filling line to the start of the next layer, where it'll start with printing support in that corner. At least the shortest route while avoiding hitting the object and the support. You can turn off Avoid Printed Parts While Travelling to cross straight through without regard for hitting the object. You'll get a few more blips then, so I wouldn't recommend it. You can turn off Avoid Supports When Travelling to make it cut across the support, which may shave off a few millimetres from the travel but can cause the support to topple over in some cases.

So the travel move is still expected (not a bug), and I think, important (but that is up to the creator of that profile).

closed 2020-08-05 by Ghostkeeper

Anon0ne, 2020-08-05:
Actually, it's the blue line that's all over in that corner. at the end of that particular piece; but sure. I have switched slicers for now and plan to revert to older Cura.

Ghostkeeper, 2020-08-05:
The dark blue line in that corner is a number of travel moves, each travelling from one Support Brim line to the next. It's done at the beginning of the layer while printing the support brim.

There is also a travel move from the corner of the build plate to the innermost Support Brim contour which cuts right through everything. This is the very first travel move it shows. It starts with the innermost Support Brim contour because working from inside to outside spreads out any blips rather than bunching them up in the inside.

There is also a travel move from the support infill pattern there off to the big Support Interface part, which is again the shortest route from the end of that last infill line to the contour of that Support Interface part (unless you disable Avoid Supports When Travelling).

I think maybe the issue is that the travel moves are too complex to understand, then? So far it looks like it's all necessary travels.

FiveOneEcho, 2020-12-06:
I just want to drop by to say this issue actually appears to do with "wipe between layers".  It happens exactly as stated and I really wish it wasn't such a shitshow reading through this.