Ultimaker/Cura #8139: "Fill gaps between walls" does not use retraction.

Stormwing01, 2020-08-02:
**Application version**
4.6.1 (i think)

**Platform**
Windows 10 pro

**Printer**
Custom printer made for my printer, because the custom nozzle size in machine settings is STILL missing if I select ender 3. (I can only select nozzles in the main menu, and the listed nozzle sizes are not what I use. )
I use a 0.35mm nozzle, and my walls are 0.18mm for outer and 0.25mm inner.

**Reproduction steps**
I made a lithophane. I imported a high detail picture and set the smoothing to zero. the base was set to 0.3mm and the thickness is set to 1.8mm. I set "fill gaps between walls" to "everywhere". I am printing with 6 walls and 95 percent infill (grid)
if you look at the travel moves, the travel moves between the filled gaps in the walls are NOT using retraction. this causes a LOT of blobbing OUTSIDE the walls thanks to oozing.

the "retraction minimum travel" doesn't work because it refuses to change anything below 0.4mm

**Screenshot(s)**

Here are two examples. light travel lines are with retraction. dark blue ones are without retractions.

[image]
[image]

**Actual results**
Fill gaps between walls does not use retraction at all between travel moves where the spaces between walls are filled.

**Expected results**
it should use retraction EVERYWHERE. (that's why I enabled my retraction afterall...)

**Project file**
You can see this with any lithophane with a lot of details.

**Log file**
N/A

**Additional information**
N/A

smartavionics, 2020-08-02:
Could you please provide the project file? Do File -> Save and rename or zip the .3mf file so you can attach to this issue. Thanks.

Stormwing01, 2020-08-02:
> 
> 
> Could you please provide the project file? Do File -> Save and rename or zip the .3mf file so you can attach to this issue. Thanks.

here's the zip with the 3mf file. not exactly a good lithophane, but if you look at the sharp bits, you can see that there are many travels without any retraction which REALLY messes up the lithophanes

[image]

[New WinRAR ZIP archive.zip](https://github.com/Ultimaker/Cura/files/5012531/New.WinRAR.ZIP.archive.zip)

Stormwing01, 2020-08-02:
I just updated from 4.6.1 to 4.6.2 and the issue persists.  the program doesn't retract with ALL travels, and only some. retraction minimum travel still doesn't seem to do anything at all below 0.4mm, because retractionless travels persist if they are small travels.

smartavionics, 2020-08-02:
By setting Maximum Retraction Count to a higher number than default and turning off combing, I get zero un-retracted travel. I wonder if the max retraction count is actually working correctly, seems odd that the default value (90) didn't work for this print.

[image]

Stormwing01, 2020-08-02:
> 
> 
> By setting Maximum Retraction Count to a higher number than default and turning off combing, I get zero un-retracted travel. I wonder if the max retraction count is actually working correctly, seems odd that the default value (90) didn't work for this print.
> 
> [image]

I want to freaking hug you right now and slap myself...  seems my maximum retraction count was WAY too low... setting that to 900 as you have it has cleared up my litho substantially (save for some tiny coasting lines, which i'm fine with... a little minimum travel too (0.5mm) is something im fine with too, now that I know the issue.

Thanks again! and a lot.
[image]
[image]

smartavionics, 2020-08-02:
Yeah, leave this issue open so that a UM dev can comment on whether the retraction count mechanism is performing as it should.

Stormwing01, 2020-08-02:
> 
> 
> Yeah, leave this issue open so that a UM dev can comment on whether the retraction count mechanism is performing as it should.

Will do.

Ghostkeeper, 2020-08-04:
I see two settings that you'll most likely want to adjust if you really want retractions everywhere.

One is already named, the maximum retraction count. The problem there is that it limits it currently to 90 retractions per 2.7mm of filament. 90 retractions is a lot, but since these gap filling walls and the lines around it are so short and small, and due to your 0.04mm layer height and small line width, that 2.7mm of filament goes a long way.
For illustration, a calculation: 2.7mm of filament for your printer is about [6.5mm³](https://www.wolframalpha.com/input/?i=pi+*+%281.75mm+%2F+2%29%5E2+*+2.7mm) of plastic. At a layer height of 0.04mm, with your layers being about 2.8mm by 60mm rectangles, that ends up at [approximately 1 layer](https://www.wolframalpha.com/input/?i=pi+*+%281.75mm+%2F+2%29%5E2+*+2.7mm+%2F+2.8mm+%2F+60mm). I've counted retractions on layer 411 of this print (seemed to be about the worst one). It did 33 retractions during the printing of the walls because of many small pieces of inner wall and then 67 during the printing of the small gaps, so that seems about right. It's thinking that it needs to protect your filament from grinding down due to too many retractions.

If you want all travel moves to be retracted, you also need to disable combing. It's currently moving through the infill area a lot while filling small gaps. These don't need to be retracted because oozing on the inside is normally not a problem. But if you do consider that a problem you set Combing Mode to "Off".

You probably don't want to adjust Retraction Minimum Travel any further. But setting that to 0 then makes it retract everywhere:
[image]

To me this seems to be behaving as expected then. I don't think making it retract everywhere by default is a good idea, but that is really up to the testers of those profiles you're using.

closed 2020-08-04 by Ghostkeeper

Stormwing01, 2020-08-04:
> I see two settings that you'll most likely want to adjust if you really want retractions everywhere.
>
> One is already named, the maximum retraction count. The problem there is that it limits it [...]

That is a VERY good explanation, and I thank you for it!