Ultimaker/Cura #8141: [4.6.2] Print profiles not showing

D-Dub24, 2020-08-03:
**4.6.2**
**Windows 10 Pro, I7-9700, 16GB RAM, GTX 1070 Ti **
**Artillery Sidewinder X1, 3D Nexus profile**

**Reproduction steps**
1. Imported machine profile and four print profiles added into appropriate folders
2. Opened cura, added Artillery Sidewinder X1
3. Imported four custom profiles via preferences/profiles
3.5. Open print settings window, and select profile drop down (should be four profiles, shows all four profiles)
4. Exit cura
5. Open Cura
6. Select drop down for profiles to change and only the last used profile which is already selected is available.
7. If I add another printer (Ultimaker S5) select the profile drop down.
8. Switch back to the artillery machine
9. Select profile drop down, and all four imported profiles show again.
10. Exit Cura
11. Open Cura, profiles are gone again except the last used profile.

**Screenshot(s)**
[image]
[image]
[image]

**Actual results**
I am unable to change profiles unless the above actions are completed in the order above.

**Expected results**
I did not have this issue in 4.6.1, and I am going to uninstall and go back to using that until this is fixed.

**Project file**
N/A

**Log file**
No Crashes

**Additional information**
I have tried to uninstall the program and reinstall multiple times without any change.
I have also tried to delete all machine and print profiles except for the ones I want, and it makes things worse, as I am not able to do my work around if I delete other machine profiles.

nallath, 2020-08-03:
Could you still add the logfile as per the template? It doesn't matter that there wasn't a crash, the log file will still show information that we need.

Onos86, 2020-08-03:
I'm having a similar issue, printer is an ender 3 pro if that matters at all. When I try to import a profile it says succesful, but the profile doesn't show up under custom profiles, restart cura and still not there

Onos86, 2020-08-03:
Here's the gcode I'd like to make into a profile, it printed nicely compared to my other test prints i did of the same stl but can't remember all the settings I used to remake the profile
[molleclip.zip](https://github.com/Ultimaker/Cura/files/5014620/molleclip.zip)

Ghostkeeper, 2020-08-04:
@Onos86 see https://github.com/Ultimaker/Cura/issues/8080.

D-Dub24, 2020-08-05:
> information
Tried to get you a log file but not sure how to do that.
How do I get you the log file?

Ghostkeeper, 2020-08-06:
Instructions are here: https://github.com/Ultimaker/Cura#logging-issues

The log file is located in `C:\Users\<your username>\AppData\Roaming\cura\4.6\cura.log`

However I recently got an issue reported that [linked to the profiles you are mentioning](https://www.reddit.com/r/Cura/comments/i30il1/some_walls_not_rendering_trying_to_print_nintendo/g0hj4je/). These profiles are made for Cura 4.6.1. It's a bit unusual, but this time profiles made for 4.6.1 are not compatible with Cura 4.6.2. This is why the profiles are not showing up.

Profiles normally get updated automatically by Cura to become compatible with the version you're using. However that doesn't hold for profiles that you manually copied into the resources folder. You've essentially installed a mod that is not compatible with your Cura version. Sorry! Please use 4.6.1 that the mod was built for.

closed 2020-08-06 by Ghostkeeper

D-Dub24, 2020-08-06:
Oh, strange. Thank you for your time.