Ultimaker/Cura #8145: Tree support and support blocker mesh es don't play well together

pietchaki, 2020-08-04:
**Application version**
4.6.1

**Platform**
Windows

**Printer**
Generic RepRap Graber i3 (Prusa i3 alike)

**Reproduction steps**
Problem is that sometimes tree support is not generated where I need it (high overhang angle, when I set it not to be generated somewhere else.
As an example:
> Branch angle:50º 
> Branch Distance: 1mm
> Branch Diameter: 1mm
> Support Overhang Angle: 70º
Support is generated correctly

If now I put a "support blocker" in the eye, where there was a support branch before, no support is generated under the chin, where it used the same "tree" but it is still needed.

**Screenshot(s)**
[groot.zip](https://github.com/Ultimaker/Cura/files/5024567/groot.zip)
[image]

**Actual results**
The entire tree is removed as one branch was inside a support blocker mesh. This does not always happen, there are some times that only the branches touching the blocker are removed. Usually, this happens when the bigger branch is removed, maybe it's the "main branch" of that tree?

**Expected results**
Only the branches inside the support blocker mesh should be removed, leaving other branches of the same tree untouched.
In the example, should generate support for the chin area, as before, but not in the eye.

**Project file**
groot_supports_70.3mf is where support was generated correctly.
groot_supports_withBlocker.3mf is using the blocker mesh, ant no tree was generated.
[groot.zip](https://github.com/Ultimaker/Cura/files/5024567/groot.zip)

**Log file**
[cura - Copia.log](https://github.com/Ultimaker/Cura/files/5024644/cura.-.Copia.log)

**Additional information**
It happens in other branches too, as well as in other files.

Ghostkeeper, 2020-08-05:
Hmm, that's not happening for me.
[image]

The mesh you shared is a plain old 3MF. Could you do a File -> Save... and save the actual project file? It'll contain the exact settings you're using too.

Could be that it's also fixed in the meanwhile. We've made some fixes for tree support in 4.7, and I'm using the alpha version that will be released as a beta soonish.

pietchaki, 2020-08-07:
Oh, I used the Export instead of Save... 
Can confirm that in the 4.7 Beta version this does not happen. Also the generated trees are really different from the ones in 4.6. It may have fixed this issue.

Anyway, if you're interested, here are the correct files from 4.6:
[groot.zip](https://github.com/Ultimaker/Cura/files/5043347/groot.zip)

I do see another problem now in 4.7: With almost the same settings as in your screenshot, just "Support placement: Touching Buildplate", the first layers of the chin have no support, also there's an area on his left arm without supports, even with plenty room to place them on the buildplate. (3_groot_supports_touching-buildplate.3mf in the previous file)
I could "fix" the chin area by decreasing the "Support overhang angle" to 65º, but the arm keeps unsupported. These areas are printing on air, so they should be supported anyway right? or I'm missing something here?
Also, should I move this one to a new issue report?

Ghostkeeper, 2020-08-10:
I think it's best to create a separate issue. We won't find new issues you're reporting here by searching. An application or indeed a feature like this is never really bug-free, so posting related issues in the same bug report leads it to being open until hell freezes over. I'll close this as the issue with blocker meshes seems to be fixed.

Sometimes it seems like the tree support branches accidentally move inside the model. The entire branch is then removed when using Buildplate Only. Maybe something is still going wrong there.

closed 2020-08-10 by Ghostkeeper