Ultimaker/Cura #8147: [master] - random crashes when loading project files

smartavionics, 2020-08-05:
Every now and again, I get a crash when loading a project file. Just happened now with the attached project file. Started Cura, loaded project file, bang...

```
2020-08-05 07:32:33,418 - INFO - [MainThread] 3MFReader.ThreeMFWorkspaceReader._processQualityChanges [816]: Loading custom profile [SV01_standard] from project file
2020-08-05 07:32:33,934 - DEBUG - [MainThread] 3MFReader.ThreeMFWorkspaceReader.read [767]: Workspace loading is notifying rest of the code of changes...
2020-08-05 07:32:33,935 - INFO - [MainThread] cura.UI.MachineActionManager.addDefaultMachineActions [74]: Default machine actions added for machine definition [creality_cr10]
2020-08-05 07:32:33,937 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [189]: Extruder [creality_base_extruder_0 #24] has already been added to this stack [SV01_mb]
2020-08-05 07:32:33,968 - INFO - [MainThread] cura.CuraApplication._setLoadingHint [774]: Initializing Active Machine...
2020-08-05 07:32:34,280 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2020-08-05 07:32:34,364 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2020-08-05 07:32:34,383 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2020-08-05 07:32:34,385 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
/opt/cura/lib/cura/plugins/3MFReader/ThreeMFReader.py:117: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = numpy.fromstring(savitar_node.getMeshData().getFlatVerticesAsBytes(), dtype=numpy.float32)
2020-08-05 07:32:34,615 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [552]: Calculating normals took 0.0003502368927001953 seconds
2020-08-05 07:32:34,745 - DEBUG - [MainThread] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0037758350372314453 seconds. 100 input vertices. 100 output vertices.
2020-08-05 07:32:34,754 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [552]: Calculating normals took 0.0003323554992675781 seconds
2020-08-05 07:32:34,838 - DEBUG - [MainThread] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.003558635711669922 seconds. 100 input vertices. 100 output vertices.
2020-08-05 07:32:34,863 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [552]: Calculating normals took 0.0006201267242431641 seconds
2020-08-05 07:32:34,874 - DEBUG - [MainThread] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.009690284729003906 seconds. 149 input vertices. 149 output vertices.
2020-08-05 07:32:34,880 - INFO - [MainThread] UM.Backend.Backend._onSocketError [218]: Backend crashed or closed.
2020-08-05 07:32:34,880 - DEBUG - [Thread-4] UM.FileHandler.ReadFileJob.run [71]: Loading file took 2.4 seconds
2020-08-05 07:32:34,882 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2020-08-05 07:32:34,887 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [315]: Attempting to kill the engine process
2020-08-05 07:32:34,888 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
Fatal Python error: Segmentation fault

Thread 0x00007f1c6e7fc700 (most recent call first):
  File "/opt/cura/lib/python3.5/threading.py", line 297 in wait
  File "/opt/cura/lib/python3.5/site-packages/zeroconf/__init__.py", line 2117 in wait
  File "/opt/cura/lib/python3.5/site-packages/zeroconf/__init__.py", line 1434 in run
  File "/opt/cura/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/opt/cura/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f1c6effd700 (most recent call first):
  File "/opt/cura/lib/python3.5/threading.py", line 297 in wait
  File "/opt/cura/lib/python3.5/threading.py", line 549 in wait
  File "/opt/cura/lib/cura/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/opt/cura/lib/python3.5/threading.py", line 862 in run
  File "/opt/cura/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/opt/cura/lib/python3.5/threading.py", line 882 in _bootstrap
Thread 0x00007f1c6f7fe700 (most recent call first):
  File "/opt/cura/lib/python3.5/threading.py", line 297 in wait
  File "/opt/cura/lib/python3.5/site-packages/zeroconf/__init__.py", line 2117 in wait
  File "/opt/cura/lib/python3.5/site-packages/zeroconf/__init__.py", line 1262 in run
  File "/opt/cura/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/opt/cura/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f1c6ffff700 (most recent call first):
  File "/opt/cura/lib/python3.5/site-packages/zeroconf/__init__.py", line 1183 in run
  File "/opt/cura/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/opt/cura/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f1c87fff700 (most recent call first):
  File "/home/markb/.local/share/cura/master/plugins/RawMouse/RawMouse/RawMouse.py", line 234 in _run_hid
  File "/opt/cura/lib/python3.5/threading.py", line 862 in run
  File "/opt/cura/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/opt/cura/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f1c9d7fa700 (most recent call first):
  File "/opt/cura/lib/python3.5/threading.py", line 293 in wait
  File "/opt/cura/lib/python3.5/threading.py", line 424 in acquire
  File "/opt/cura/lib/python3/dist-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/opt/cura/lib/python3/dist-packages/UM/JobQueue.py", line 123 in run
  File "/opt/cura/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/opt/cura/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f1c9dffb700 (most recent call first):
  File "/opt/cura/lib/python3.5/threading.py", line 293 in wait
  File "/opt/cura/lib/python3.5/threading.py", line 424 in acquire
  File "/opt/cura/lib/python3/dist-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/opt/cura/lib/python3/dist-packages/UM/JobQueue.py", line 123 in run
  File "/opt/cura/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/opt/cura/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f1c9e7fc700 (most recent call first):
  File "/opt/cura/lib/python3.5/threading.py", line 293 in wait
  File "/opt/cura/lib/python3.5/threading.py", line 424 in acquire
  File "/opt/cura/lib/python3/dist-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/opt/cura/lib/python3/dist-packages/UM/JobQueue.py", line 123 in run
  File "/opt/cura/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/opt/cura/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f1c9effd700 (most recent call first):
  File "/opt/cura/lib/python3.5/threading.py", line 293 in wait
  File "/opt/cura/lib/python3.5/threading.py", line 424 in acquire
  File "/opt/cura/lib/python3/dist-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/opt/cura/lib/python3/dist-packages/UM/JobQueue.py", line 123 in run
  File "/opt/cura/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/opt/cura/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f1cce4a8700 (most recent call first):
  File "/opt/cura/lib/python3/dist-packages/UM/Event.py", line 212 in __init__
  File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 210 in emit
  File "/opt/cura/lib/python3/dist-packages/UM/Settings/ContainerStack.py", line 802 in _emitCollectedPropertyChanges
  File "/opt/cura/lib/python3/dist-packages/UM/Event.py", line 218 in call
  File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 464 in event
  File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 1060 in event
  File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 844 in run
  File "/opt/cura/bin/cura", line 249 in <module>
Segmentation fault (core dumped)
```

[CCR10_magni_combo_mount_v2.zip](https://github.com/Ultimaker/Cura/files/5026547/CCR10_magni_combo_mount_v2.zip)

Ghostkeeper, 2020-08-06:
It doesn't happen every time, you say? Not even with the same project file?

I'm able to open that project file fine, and haven't seen this problem yet so far.
[image]

Maybe it's got something to do with the Qt version or something? Which version were you using?

Ghostkeeper, 2020-08-06:
It could also have something to do with your [other recent issue with project loading](https://github.com/Ultimaker/Cura/issues/8134) which also hints towards race conditions.

smartavionics, 2020-08-06:
No, it doesn't happen every time, just once in a while. As you say, this and the other issue do look racy. I have a suspicion that project loading is more likely to fail if you load immediately after the UI is displayed while there's various threads loading machine defs, etc. No proof of that.