Ultimaker/Cura #8149: Add-ons

drg10350, 2020-08-05:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

drg10350, 2020-08-05:
Re Cura 4.6.2:

Have just downloaded latest, and noticed:

 there are no Mesh Fix tools as in previous version

Ability to set own printer prefix per printer has gone

Help!

Many thanks

David

closed 2020-08-05 by drg10350

fieldOfView, 2020-08-06:
I am not sure if you meant to close your issue, but both things you mention are plugins that may have to be reinstalled: MeshTools and Custom Printjob Naming.

drg10350, 2020-08-06:
Many thanks - and no I did not - very careless. Solution was staring me in the face but old age forgot....

David