Ultimaker/Cura #8150: Allow models being moved on positive Z

Liger0, 2020-08-05:
**Is your feature request related to a problem? Please describe.**
Some items downloaded are supposed to be put together for multi-color printing or "chaining", but fail to do so because of being exported incorrectly. So you could still move manually them in the air to pose them, but the model can only touch the bed

**Describe the solution you'd like**
Allow the model to be moved up on the Z, with a checkbox of "touch the build plate" being optional

videopilot, 2020-08-06:
Doesn't turning off the "automatically drop to build plate" preferences option solve this?

Liger0, 2020-08-06:
> 
> 
> Doesn't turning off the "automatically drop to build plate" preferences option solve this?

I don't see that option anywhere, where is it?

Ghostkeeper, 2020-08-06:
It's in the preferences window:
[image]

Ghostkeeper, 2020-08-06:
It's enabled by default because otherwise people have lots of failed prints because their models are floating in mid-air.

Liger0, 2020-08-06:
I understand that. It would be useful to have a per-item one anyway to make it selectable when moving it without. Like you can move a piece in the air but the new ones imported will still drop automatically on the build plate.

videopilot, 2020-08-06:
Well, when you turn that option off, new models still import on the build plate. Anything you then move up or down can be placed back to the build plate by typing 0 for its Z position. Seems complicated to make some objects moveable up and down and not others.

Ghostkeeper, 2020-08-07:
I'll close this as duplicate of https://github.com/Ultimaker/Cura/issues/8117 then, where we discuss about overriding this preference for certain models.

closed 2020-08-07 by Ghostkeeper