Ultimaker/Cura #8155: Brim lines cut off with normal supports

Stormwing01, 2020-08-06:
**Application version**
cura 4.6.2

**Platform**
Windows 10 pro

**Printer**
Custom printer.

**Reproduction steps**
I have included the 3mf file. whenever there are smaller spots on the build plate and I activate supports, the brim gets cut off. the supports make it so that only 1 brim line is made, then the supports, and then the rest of the brim. so in effect, there is no brim holding the print down. this only seems to be the case for normal supports, not tree supports.

**Screenshot(s)**
[image]

**Actual results**
the results are explained in the reproduction steps

**Expected results**
a full brim that should support the print

**Project file**
[New WinRAR ZIP archive.zip](https://github.com/Ultimaker/Cura/files/5036586/New.WinRAR.ZIP.archive.zip)

**Log file**
N/A

**Additional information**
support brim is disabled, but there is nothing different when it's enabled or disabled.

Ghostkeeper, 2020-08-07:
I see it happen, indeed. It seems like the "Brim Replaces Support" setting works on support, but not on support interface. This has probably been an oversight in the implementation there.

Vandrasc, 2020-08-18:
Fixed in the stable release, thank you for your feedback!

closed 2020-08-18 by Vandrasc