Ultimaker/Cura #8156: Linear Advance as a Profile Setting

Anon0ne, 2020-08-06:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])
Not really, just have to modify the machine start code for different filaments.

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)
Allow us to set the Linear Advance in "Advanced Options" for each profile.. Hidden from other users by default.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)
I would need multiple machine's added to list with separate start codes.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)
Anyone running linear advance who changes K factor for material settings.

**Additional context**
(Add any other context or screenshots about the feature request here.)

fieldOfView, 2020-08-06:
There is a plugin for this in the Marketplace.

dankatanka, 2020-08-10:
@fieldOfView this plugin broke saving file or print via octorpint for a long time. I use this plugin with fix:
```py
            gcode_command_pattern = "M900 K%f T%s"
```

fieldOfView, 2020-08-10:
@dankatanka, thanks for reporting that, but please use the LinearAdvanceSetting issue queue next time (and please don't wait "for a long time" ;-) ). https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/issues

Ghostkeeper, 2020-08-10:
We're not in a hurry to implement something that is made available through the community. Not to say that functionality from such plug-ins are forbidden terrain and should never be merged into mainline Cura. But it's at least less pressing if a good solution exists like this.